RaspAP/raspap-webgui #121: Translation

Markus1995, 2017-10-17:
Hello,

Can You Please integrate a Translation function or Script?
I can help you to translate.

billz, 2017-10-17:
It's a good suggestion. Which language could you contribute?

bebora, 2017-10-22:
I can contribute to the Italian translation if you want.

Markus1995, 2017-10-22:
I can translate into German

topataco, 2017-10-24:
I can help translating into Spanish if needed

billz, 2017-10-25:
@Markus1995 @bebora @topataco thanks for your kind offers to help translate RaspAP!

The process is, I think, pretty straightforward. Start by downloading the Portable Object file  [messages.po](https://raw.githubusercontent.com/billz/raspap-webgui/i18n/locale/en_US/LC_MESSAGES/messages.po) and replace the `msgstr` messages with your translations. For example, an original English `msgid` and the French `msgstr` translation is below:
```
msgid "Dashboard"
msgstr "Tableau de bord"
```
Please be sure to include accent characters, if applicable. Email the translated .po files back to me. I'll compile them into Machine Object (.mo) files and add them to the project. You'll be credited as the original translators.
 
Many thanks.

billz, 2017-10-29:
Italian translation is looking good. Thanks @bebora 
 
[image]

Markus1995, 2017-10-29:
I will translate it tomorrow.
Thank You for the opportunity

ImEmil, 2017-11-21:
I'm planning on installing this when I get my raspberry pi this week.
I can translate this into Swedish @billz

billz, 2017-11-21:
@ImEmil thanks, it would be a nice addition.

ImEmil, 2017-11-22:
@billz Check your gmail and thanks for this awesome package.

dalmago, 2017-12-07:
I can translate into portuguese (Brazil)

billz, 2017-12-07:
@ImEmil I received your email, many thanks.
@dalmago that would be outstanding.

I'll rebase the i18n branch and include your translations

dalmago, 2017-12-11:
@billz Did you get my email as well?

billz, 2017-12-13:
@dalmago got it, thank you!

billz, 2018-05-28:
I've merged #125 which includes a basic framework for GNU locale support, along with two translations: French (mine) and Italian (many thanks @bebora). A native French speaker could probably improve on my work 😅 

@dalmago @ImEmil I received your .PO files. Again, thank you. Some new strings have been added to the UI since then. Would you kindly translate these as well? https://pastebin.com/raw/6g9HkfBU

@topataco @Markus1995 Spanish and German would round up our translations nicely. Here's a link to the current [Portable Object file](https://raw.githubusercontent.com/billz/raspap-webgui/master/locale/en_US/LC_MESSAGES/messages.po) for the project. 

I'll update the project wiki with a translation page shortly.

ImEmil, 2018-05-29:
@billz I've emailed you the additional strings translated to Swedish 👍

billz, 2018-06-15:
sprechen Sie Deutsch? Big thanks to Tino Undeutsch for his German translation 🇩🇪  

[image]

BANKA2017, 2018-10-15:
@billz  I will translate it tomorrow

BANKA2017, 2018-10-16:
@billz I updated the file [messages.po](https://gist.github.com/BANKA2017/05346a616fca35e600019a05182ab907/) , the additional strings translated to Chinese simplified

billz, 2018-10-17:
@BANKA2017 I've added support for Chinese simplified with your translation. thank you! it's a great addition to the project 🇨🇳

[image]

rvallo, 2018-11-03:
@billz I've uploaded you the additional strings translated to Czech :+1:  [messages.po](https://gist.github.com/rvallo/6a915e8d84aad862c348b918e1114ab2)

billz, 2018-11-04:
@rvallo fantastic! Thank you for this contribution. Great to have Czech supported by the project 🇨🇿

[image]

billz, 2018-11-19:
Say 'Привет!' to RaspAP in Russian. Many thanks Betep for the translation 🇷🇺

[image]

billz, 2019-03-04:
Added two new translations to the project: Finnish and Sinhala. Many thanks to M4ra and Damith, respectively! 🇫🇮 🇱🇰 

[image]

[image]

billz, 2019-06-10:
Added Turkish language support to the project. Many thanks Cuqer! 🇹🇷 

[image]

billz, 2019-10-22:
If you previously contributed a translation to the project (or wish to get involved) please see #418. Thanks

billz, 2019-10-29:
All existing translations have now been updated...and support for 4 new languages has been added to the project 🎉 

RaspAP is now available in Indonesian, 한국어 (Korean), 日本語 (Japanese) and Tiếng Việt (Vietnamese). 🇮🇩 🇰🇷 🇯🇵 🇻🇳  Details [here](https://github.com/billz/raspap-webgui/wiki/Translations#supported-languages).

billz, 2020-01-14:
Added Greek language support to the project. Many thanks to Chris K for his translation work! 🇬🇷 

[image]

billz, 2020-04-05:
The project has a new translation home at Crowdin https://crowdin.com/project/raspap
This is a public project with an open source license. All are welcome to contribute.

Will update the docs accordingly.

billz, 2020-04-10:
Pleased to announce Danish language support. Thanks to the generous contribution of Svend Skipper Andersen (skipperskip) 🇩🇰⛵️

[image]

marco97pa, 2020-04-12:
> The project has a new translation home at Crowdin https://crowdin.com/project/raspap
> This is a public project with an open source license. All are welcome to contribute.
> 
> Will update the docs accordingly.

Added more strings to the italian translation on Crowdin. I hope you will add my name to the contributors section

billz, 2020-04-12:
@marco97pa grazie! Crowdin does not update the last-translator field, so I've added attribution for you. thanks for your contribution

billz, 2020-04-15:
If Chinese traditional (正體中文) is your preferred language, say 'thanks!' to contributor @squareface0323 who did this translation in record time 🇹🇼 
[image]

billz, 2020-10-01:
Added Polish language support to the project. Many thanks to Patryk Kubiak for his translation work! 🇵🇱

[image]

billz, 2021-12-31:
Say 'bună ziua!' to RaspAP in its newest language — Romanian 🇷🇴 Many thanks to Drago for his translation work!

[image]

Minobi, 2024-08-13:
I can improve Russian translation if you interested in this. Now there are some typos and empty strings and a little incorrect translation.

billz, 2024-08-13:
@Minobi that would be great, thanks. you can find the project here https://crowdin.com/project/raspap