RaspAP/raspap-webgui #126: wpa_cli always return "command timed out"

ami4TECH, 2017-10-25:
Hi..
In the "Configure Client" page all functions using wpa_cli are returning "command timed out".
see output of the var_dump below
--------------------------------------------
  exec( 'sudo wpa_cli scan', $MyRest );
  sleep(3);
  exec( 'sudo wpa_cli scan_results',$scan_return );
echo "<br>===========<br>";
var_dump($MyRest);
echo "<br>===========<br>";
var_dump($scan_return);
echo "<br>===========<br>";
--------------------------------------------

output
--------------------------------------------
===========
array(2) { [0]=> string(34) "Selected interface 'p2p-dev-wlan0'" [1]=> string(25) "'SCAN' command timed out." } 
===========
array(2) { [0]=> string(34) "Selected interface 'p2p-dev-wlan0'" [1]=> string(33) "'SCAN_RESULTS' command timed out." } 
===========
[image]

--------------------------------------------

The commands working fine from terminal perfectly.
RPi3
PHP Version 7.0.19-1
Apache 2.0

billz, 2017-10-25:
Did you add the www-data user to the sudoers file?

ami4TECH, 2017-10-25:
Yes, I did.
BTW, I had to add the line below as well
www-data ALL=(ALL) NOPASSWD:/sbin/ip

Thank you,

billz, 2017-10-25:
Did you also include the following? 
```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
```
In most cases timeouts like these occur when permissions aren't set correctly. Since you're using Apache, rather than Lighttpd with the installer, I assume you followed the manual configuration steps?

ami4TECH, 2017-10-26:
Yes, I flowed the manual configuration steps
below is all that i have included in the sudoers

www-data ALL=(ALL) NOPASSWD:/sbin/ifdown wlan0
www-data ALL=(ALL) NOPASSWD:/sbin/ifup wlan0
www-data ALL=(ALL) NOPASSWD:/bin/cat /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/wifidata /etc/wpa_supplicant/wpa_supplicant.conf
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/hostapddata /etc/hostapd/hostapd.conf
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd stop
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq stop
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/dhcpddata /etc/dnsmasq.conf
www-data ALL=(ALL) NOPASSWD:/sbin/shutdown -h now
www-data ALL=(ALL) NOPASSWD:/sbin/reboot
www-data ALL=(ALL) NOPASSWD:/sbin/ip

SirLagz, 2017-10-26:
You're missing some lines, that's why you had to add the /sbin/ip. I forgot to update the manual install instructions when I updated the install script. whoops!

ami4TECH, 2017-10-26:
Same thing in using webconsole 

sudo wp a_cli scan

**Selected interface 'p2p-dev-w
lan0'
'SCAN' command timed out.**

ami4TECH, 2017-10-29:
The issue seems to be related to fastcgi-php.
I could not enable fastcgi in Apache.
I'm using lighttpd now and "Configure Client" is working fine.

I have added/update couple of lines in dashboard.php and I'll list the changes below in case it be of any use.
My SSID contains "-" in the name and dashboard was reporting "Not connected" in the network name, to fix this line 38 changed to
**preg_match( '/ESSID:\"([a-zA-Z*-9\s]+)\"/i',$strWlan0,$result ) || $result[1] = 'Not connected';**

Stop and start wlan0 button was not refreshing the page after it finished and was not reporting the current stats of the interface
The line
      **exec( 'sudo ip link set wlan0 down',$return );**
Changed to
      **exec( 'sudo ip link set wlan0 down',$return );
      sleep(3);
      header('Location: '.$_SERVER['REQUEST_URI']);**

And lines
      **exec( 'sudo ip link set wlan0 up',$return );
      exec( 'sudo ip -s a f label wlan0',$return);**
Changed to
      **exec( 'sudo ip link set wlan0 up',$return );
      exec( 'sudo ip -s a f label wlan0',$return);
      sleep(3);
      header('Location: '.$_SERVER['REQUEST_URI']);**

Sorry, i did not use github before and I'm not familiar with reporting code updates best practice.

Thank you for the great work

SirLagz, 2017-10-29:
@ami4TECH Thanks for these. We'll make sure your work gets integrated into the main branch.

billz, 2019-04-13:
Marking as resolved.

closed 2019-04-13 by billz

davidfischer, 2020-02-14:
I stumbled across this issue while debugging a similar problem. The reason for this is that Raspbian's Apache systemd setup specifies `PrivateTmp=true` in `/lib/systemd/system/apache2.service` while lighttpd does not. It seems some of the communication between `wpa_cli` and `wpa_supplicant` relies on the regular temp directory.

luke-pefi, 2020-02-27:
@davidfischer Thank you so much for your comment!  I was at my wits end `why wpa_cli` scan would time out only if run from Apache.  Saved me a ton of debugging.

sctse999, 2020-05-26:
@davidfischer You save my day!  I wouldn't think of PrivateTmp will be the reason