RaspAP/raspap-webgui #127: Fix non-Debian's way of configuring stuff

michalfita, 2017-10-26:
Thanks for very good project - it's very useful. However, because my installation failed I started to dig how you configure things.

I figured out many of your changes actually do harm to the delicate fabric of Debian's (modern SYSV style) configuration structure. Below I present suggestions how to improve things to reduce _damages_ and potential problems on updates of the system itself:

1. Changes in `/etc/sudoers` shall go to `/etc/sudoers.d/090_raspap` (that's mentioned in #41 already I belive)
2. Changes in `/etc/dnsmasq.conf` shall go to `/etc/dnsmasq.d/090_raspap.conf`
3. Changes in `/etc/rc.local` shall not land there at all:
   - The enable of IP forwarding shall go to `/etc/sysctl.d/90_raspap.conf` with the line `net.ipv4.ip_forward = 1`.
   - The configuration of the iptables shall
      - land in separate script in `/etc/init.d/` symlinked to `/etc/rc.d/S99raspap`; or
      - use systemd, which is harder (refer [here](https://jstrapko.github.io/systemd-iptables/) for help)

I understand some of them are applicable only to the recent version of Raspbian/Debian, but it would be nice to take _correct_ way into account.

I implemented above changes manually and they seem fine (except dnsmasq).

I would be happy to provide a pull request with these changes, hover at least one of it required digging in your PHP scripts, what I would prefer to avoid in my limited time (I already spent half a day sorting things out manually while I have tests to perform on the product I'm working on).

SirLagz, 2017-10-26:
Hi @michalfita, 

Thanks for bringing these to our attention. I've already been slowly updating the install scripts to take these changes into account. The sudoers file was indeed mentioned already, and I've been looking at how to rewrite iptables part of the configuration as there needs to be something to enable/disable the iptables rules when required, and also possible adding a web interface to the iptables rules.

If you'd like, make a pull request against my personal branch @ https://github.com/SirLagz/raspap-webgui and I'll modify the code as needs be and merge them into the main branch as time permits.

SirLagz, 2017-10-27:
Had a look at /etc/sysctl.d/99-sysctl.conf and the ip forwarding rule is already there on line 28.
The installer should probably just uncomment it rather than adding another file to sysctl.d needlessly.

michalfita, 2017-10-27:
In fact the `/etc/sysctl.d/99-sysctl.conf` is a symlink to the `/etc/sysctl.conf`. The whole idea of `/etc/*.d/` directories is about avoiding edition of configuration files when packages are installed and uninstalled. The presence of the file makes it clear what in the system request particular change in the configuration.

SirLagz, 2017-10-28:
Ah that is true, I didn't notice that before.

Not sure whether you mean "addition" or "editing" of files when you said "edition", but I'm assuming you mean "editing" in which case, I agree, and have been looking at using the "Debian Way" of configuring things for a while, but haven't had the time to actually do it.

SirLagz, 2017-10-28:
After looking at ArchLinux and Pidora, we may need to build some logic to pick up the distro used, and where to save files. I'm not too familiar with ArchLinux or Pidora so unsure if they use the same `dnsmasq.d` or `sysctl.d` directories that Raspbian uses. Will need to look further into this.

jrmhaig, 2017-10-28:
I'm coming back to look at this project after being away for a while. I tried installing on a completely new Raspbian Stretch image and the access point would not start (and gave no errors). Are you fixing this? If so, I will stop digging and have a look at your branch instead.

@michalfita Thanks for the notes. I think this project started when Raspbian was still based off Wheezy and a lot of things have changed since then.

SirLagz, 2017-10-28:
I haven't gotten to fixing the wireless bit yet. I'm currently writing a GUI to change IP addresses / DNS servers etc for both wlan and ethernet devices. This ties in with the access point however.

I installed it on Stretch 9.0 and had some limited success. I think 9.1 changes a few things that may have broken RaspAP?

michalfita, 2017-10-31:
@SirLagz Pidora is Fedora/RedHat, so they probably do a lot of stuff via `systemd`. If Pidora uses Network Manager, then things are getting complicated (f.e. `/etc/NetworkManager/dnsmasq.d/`; _untested_).

I have absolutely no experience with ArchLinux, I never even stand nearby.

SirLagz, 2017-10-31:
@michalfita Sounds like we'll need some testing before we can change the way we do our configuration scripts. I'll test out other distros when I get a chance.

commit 4b30be5c (2020-03-20): Align installer w/ Debian's modern SYSV style. Resolves #127
closed 2020-03-29 by billz