RaspAP/raspap-webgui #137: Stretch

ashah7, 2017-11-14:
Hi
I came across this and tried to install this on the Stretch image but it gave an error on install about the php5 packages being obsolete.   I then tried (as suggested here) using the manual install with the php-cgi package - the installation seems to have worked okay.
On reboot the hotspot didn't start - I know that in Stretch the networking has been changed.
Does webgui currently working with Stretch or is there more development needed?
thanks

billz, 2017-11-16:
Resolved with #124. Please try again with the latest, thanks.

ashah976, 2017-11-17:
thanks it installs no problems but some issues i've found

1.  hostapd doesnt start when configuring wlan0 for hostspot - was okay when I used wlan1 (edimax usb wifi)

2.  The scan command to configure the wifi return nothing so its not possible to configure wifi

ashah976, 2017-12-21:
I see from other comments after this that everyone believes this works fine.  
Has anyone tried to configure their wifi credentials ?  
Each time I scan the wifi network from the gui it returns nothing
Even if using the onboard wifi adapter I still cant get this to work..
I'm I missing something easy here?  
I haven't followed the prerequisites as I just need to connect to the hotspot and connect the pi to my router via wifi

I've tried the onboard wifi adapter
Edimax usb adapter
RT5370 Wireless Adapter
TL8188CUS 802.11n WLAN Adapter

thanks for any help

billz, 2018-02-11:
@ashah7 the project was originally developed for RPi's without onboard wifi, but is compatible with current hardware. @ashah976 re: wlan1, I just merged a pull request from @emmanuelgeoffray that allows users to define the wifi client interface in `config.php`. perhaps this will help.

billz, 2018-03-19:
Closing this for now. Feel free to reopen if you have additional feedback. Thanks

closed 2018-03-19 by billz