RaspAP/raspap-webgui #139: Eth and wifi running but no internet

djdive, 2017-11-19:
Hi,
I have installed the software on an clean rasp.
And as far I see everything is running OK, except that when I connect with an wifi client to the pi I have no internet.
Web interface is reachable. 
Is there an forward rule I missed?

dalmago, 2017-11-23:
This is probably a DNS issue. Try accessing some website by its IP address to see if your connection is okay. (http://216.58.222.78 should open Google).

Kirorus, 2017-12-07:
I have this problem too. It dose not work with IP (google or other servers)

SirLagz, 2017-12-07:
mon.wlan0 shouldn't be required, but double check the iptables configuration to make sure the forward rule has been added properly.

Paste the output of `iptables -t nat -L`

Kirorus, 2017-12-07:
My "iptables -t nat -L"
`Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination  `

I think this is not good))) how can i add forward rule?

SirLagz, 2017-12-07:
`iptables -t nat -A POSTROUTING -j MASQUERADE` <== type that into a root shell

Kirorus, 2017-12-07:
thank YOU!
after that it add string in iptable

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere  

how can I automate it at booting raspberry?

thank you!

SirLagz, 2017-12-07:
You can use iptables-persistent to save the rules, or alternatively you can place that line into /etc/rc.local

jhoward15, 2017-12-12:
Same issue here. Everything seems to been running properly and I can connect to the access point, but I can't browse the Internet. AP config web page works fine from the connecting device. 

I followed the iptables instructions above (mine was blank previously) and now mine is (and is persistent):
```
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere_            

```
I also have this in the HostAPD log:
```
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:e0:f3:27 and ssid "raspi-webgui"
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Only 0/20 bytes of strong random data available from /dev/random
random: Not enough entropy pool available for secure operations
WPA: Not enough entropy in random pool for secure operations - update keys later when the first station connects
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: associated
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Only 18/20 bytes of strong random data available from /dev/random
random: Not enough entropy pool available for secure operations
WPA: Not enough entropy in random pool to proceed - reject first 4-way handshake
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: disassociated
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: associated
WPA: wpa_sm_step() called recursively
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: disassociated
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 34:23:ba:08:86:f2
wlan0: STA 34:23:ba:08:86:f2 RADIUS: starting accounting session 5A2FFD66-00000002
wlan0: STA 34:23:ba:08:86:f2 WPA: pairwise key handshake completed (WPA)
wlan0: STA 34:23:ba:08:86:f2 WPA: group key handshake completed (WPA)
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 34:23:ba:08:86:f2
wlan0: STA 34:23:ba:08:86:f2 IEEE 802.11: disassociated

```

Can any one help? I've also attached the networking info from my test device
[image]

drndos, 2017-12-16:
Hi, you need to enable ipv4 forwarding, edit /etc/sysctl.conf to include this line, then reboot:
`net.ipv4.ip_forward=1`
Then if it still doesn't work execute as sudo
`iptables -t nat -A POSTROUTING -j MASQUERADE`

jhoward15, 2017-12-18:
> Hi, you need to enable ipv4 forwarding, edit /etc/sysctl.conf to include this line, then reboot:
net.ipv4.ip_forward=1

This fixed it for me - thanks so much!

itshaadi, 2018-01-16:
i have the same issue. although my internet port is eth1 not eth0. so does this make a difference or it will detect the internet port automatically ?

whatda, 2018-03-10:
So I installed this package on a fresh rasbian installation (following [this article](https://howtoraspberrypi.com/create-a-wi-fi-hotspot-in-less-than-10-minutes-with-pi-raspberry/)) and the wifi hotspot had no internet access...
I needed to uncomment the `net.ipv4.ip_forward=1` in `/etc/sysctl.conf` and reboot.
After that, I also needed to execute `iptables -t nat -A POSTROUTING -j MASQUERADE`.
Thx to @SirLagz and @drndos for pointing that out.

Now it worked! But I had to execute the above iptables command every time after reboot in order to have internet access on startup... Which sucks... I wanted an AP which I just have to power up in order to work.

So the solution to this last issue is a `sudo apt-get install iptables-persistent`, executing the iptables rule again and then attempt to save it with `sudo iptables-save > /etc/iptables/rules.v4`.
The last part gave me some weird permission issues, so I had to invoke a root shell with `sudo su -` and type `iptables-save > /etc/iptables/rules.v4`, which finally resolved my problems.

Maybe that'd be something to integrate into the installer, because - at least imo - one would expect a hotspot to have internet access and be rebootable out of the box, so to speak. The rest of the package is cool, though.

billz, 2019-03-12:
The Quick Installer was updated some time ago to persist these settings to `/etc/rc.local`, so the project brings up the AP after reboots. 

Marking this resolved. Thanks for reporting.

closed 2019-03-12 by billz