RaspAP/raspap-webgui #141: Everything works but no internet

rdjinns, 2017-12-13:
Hello everyone,

I followed the quick install, everything works fine, I can connect to the hotspot but the internet does not work.

I would really like help please and I do not know anything about Linux, besides my English is bad

thank you

agonzale34, 2017-12-15:
I have the same problem.

SirLagz, 2017-12-20:
Have you tried what I specified in Issue #139 ?

Caxton1, 2017-12-24:
Hi, I am new to GitHub so apologies if I don't know the rules of posting. Having had the same issue of not getting internet access, I referenced this document and pulled out the relevant bits and made the whole thing work for me. 
https://cdn-learn.adafruit.com/downloads/pdf/setting-up-a-raspberry-pi-as-a-wifi-access-point.pdf

I'm running stretch lite on rpi 3 with built in wifi connected by ethernet cable to a broadband router and used the quick installer to install raspAP.

Start with a fresh image of stretch-lite then run any updates and upgrades

sudo apt-get update && sudo apt-get upgrade -y

1.	Install iptables manager - when prompted, answer "yes"
sudo apt-get install iptables-persistent

2.	Edit hostapd.conf
sudo nano /etc/hostapd/hostapd.conf

	Comment out the driver if you are using rpi3 with wireless

#driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
beacon_int=100
auth_algs=1
wpa_key_mgmt=WPA-PSK
ssid=raspi-webgui
channel=1
hw_mode=g
wpa_passphrase=ChangeMe
interface=wlan0
wpa=1
wpa_pairwise=TKIP
country_code=AF

	Save and close the file

3.	Tell the Pi where to find this configuration file.

sudo nano /etc/default/hostapd

	This step shouldn't be needed but check anyway.
Find the line 
#DAEMON_CONF="" 
and change it to 
DAEMON_CONF="/etc/hostapd/hostapd.conf"

	Save and close the file.

sudo nano /etc/init.d/hostapd
Find the line
DAEMON_CONF=
and change it to
DAEMON_CONF=/etc/hostapd/hostapd.conf

4.	Configure Network Address Translation

sudo nano /etc/sysctl.conf

uncomment the line 
#net.ipv4.ip_forward=1

 so that it becomes

net.ipv4.ip_forward=1

	Save and close the file.

	Now run this command to activate immediately

sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"

5.	Run the following commands to create the network translation between the ethernet port eth0 and the wifi port wlan0

sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
sudo iptables -A FORWARD -i eth0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan0 -o eth0 -j ACCEPT

	To make this happen on reboot (so you don't have to type it every time) run 

sudo sh -c "iptables-save > /etc/iptables/rules.v4"

Hopefully this might help those with the same issue.

kcschenk, 2018-01-15:
Caxton1, Thank you this was the last part I needed to create my wlan0->wlan1 network I have referanced you in an instructable http://www.instructables.com/id/Raspberry-Pi-As-Completely-Wireless-Router/
Again thank you

billz, 2019-04-13:
There have been many updates to the project since this was opened. Please follow the project prerequisites and try the latest Quick Installer or manual installation steps.

Closing this issue for now. Feel free to update if your issue persists, thanks.

closed 2019-04-13 by billz

prampey, 2020-08-09:
This issue still persists at the time of writing this post. For some context, I've set up a Raspberry Pi Zero W in AP-STA mode and can connect to the hotspot but there is no internet.
I've had to manually enable IP forwarding in ```/etc/sysctl.conf``` and configure the IP tables as mentioned above.

Btw, this project makes network configuration so much easier and user-friendly. Kudos!

billz, 2020-08-09:
IP forwarding is included in both the Quick Installer and manual installation steps. The wiki is your friend :)