RaspAP/raspap-webgui #145: no ssid after installation

gifa75, 2018-01-06:
Hi,
I've installed raspap-webgui (with quick installation) in a clean RaspbianLite version after a uptade/upgrade.
Installation didn't give any error, but after reboot nothing happened.
No SSID and no more wifi.
ifconfig command give me this:

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 1920  bytes 128640 (125.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1920  bytes 128640 (125.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.3.141.1  netmask 255.255.255.0  broadcast 10.3.141.255
        inet6 fe80::69f:abc7:ac51:4257  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:ef:63:d0  txqueuelen 1000  (Ethernet)
        RX packets 114  bytes 7626 (7.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 63  bytes 8793 (8.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

and I think it's ok

But when I execute this command:
hostapd -d /etc/hostapd/hostapd.conf

the result is this:
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=2 op=0 soft=0 hard=0
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 2 iftype 3 (AP)
nl80211: Failed to set interface 2 to mode 3: -1 (Operation not permitted)
nl80211: Try mode change after setting interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
Could not set interface wlan0 flags (DOWN): Operation not permitted
nl80211: Failed to set interface down
nl80211: Interface mode change to 3 from 0 failed
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=2 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 2 iftype 2 (STATION)
nl80211: Failed to set interface 2 to mode 2: -1 (Operation not permitted)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x1a9c8d0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1a9c8d0)
wlan0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: AP-DISABLED
hostapd_cleanup(hapd=0x1a9e858 (wlan0))
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x1a9c8d0)
hostapd_interface_free: free hapd 0x1a9e858
hostapd_cleanup_iface(0x1a9c8d0)
hostapd_cleanup_iface_partial(0x1a9c8d0)
hostapd_cleanup_iface: free iface=0x1a9c8d0

So the problem seems to be wlan0 interface.
What should I check?

Thank you in advance
Giuseppe

jqcorreia, 2018-01-09:
First of all, great work! It really hits some pain points for my use cases.

I'm having the same problem as @gifa75 tough... 
I went through the same motions (installing through quick install and removing the configured network from wpa_supplicant) and now I can't configure as Wifi client since no networks appear in the list.

Any toughts?

gifa75, 2018-01-09:
#Hi @jqcorreia ,
the problem of empty list is due by this instructio that return no wifi networks:

sudo wpa_cli scan_results

A workaround I found is to execute this command:
sudo iwlist wlan0 scan

which, I don't know why, allows the previous one to work. I inserted that instruction directly in the php page :D

But now I have another problem. After inserting psk passprhase I'm unable to connect to internet.
I've also tried #139 but without success.

:(

billz, 2019-04-13:
Fixed with #297

closed 2019-04-13 by billz