RaspAP/raspap-webgui #146: Change port number

toqduj, 2018-01-06:
There's other competing web services running on my RPi. Any chance I can change the port number raspap works on?

zubenubi, 2018-01-11:
Im also interested in this

SirLagz, 2018-02-13:
You can change this manually, depending on which web server you're using. If you're using lighttpd only for RaspAP, then you can change the port lighttpd is listening on.

billz, 2019-11-10:
Shout out to Raspberry Pie on discord. Thanks for the feedback 👍🏼

closed 2019-11-10 by billz

GarethHards, 2021-01-17:
I also have raspAP on port 80 and think i should change it or should i look at changing apache2 virtual hosting ports rather instead, seems that raspAp may be the simplest way? if so how do i do this ?

billz, 2021-01-17:
@GarethHards read the FAQ https://github.com/billz/raspap-webgui/wiki/FAQs

GarethHards, 2021-01-17:
does raspAP actually interfere with apache2 virtual hosts on port 80 ? and occupy the same port ?

commit eb0442b1 (2019-11-10): Added support for lighttpd server port. Resolves #146