RaspAP/raspap-webgui #147: OpenVPN configuration management

randomly123, 2018-01-09:
Very good project: - please can you extend the functionality of the project to include the administration of OpenVPN files by the RaspAP webGUI project?

It has been referenced in the main Readme file in this manner:-

Optional services

OpenVPN and TOR are two additional services that run perfectly well on the RPi, and are a nice way to extend the usefulness of your WiFi router. I've started on interfaces to administer these services. Not everyone will need them, so for the moment they are disabled by default. You can enable them by changing these options in index.php:

// Optional services, set to true to enable.
define('RASPI_OPENVPN_ENABLED', false );
define('RASPI_TORPROXY_ENABLED', false );

Please note that these are only UI's for now. If there's enough interest I'll complete the funtionality for these optional admin screens.

andresdans, 2018-05-22:
+1
This feature would be appreciated.

rohitg76, 2018-08-14:
+1
Awaited..

r45635, 2019-05-04:
### **Good job to the dev team,**
my small contribution to global efforts for the community. Ported on raspberry pi3b+.

Should that help i did some changes manually to make it work, the major issue is the exec command in php invoking a sudo privilege not possible. 

as a prerequisit please make sure you have already installed an OpenVPN client.
I assume that if it works then such type of commande line should be ok : 

**enable the option for the HTM gui** as mentionned abobe (file index.php)
define('RASPI_OPENVPN_ENABLED', true );

**Create two scripts:**
_cd /etc/raspap/hostapd_
_sudo nano enableVPNClient.sh_
copy the content below and save the file
_#!/bin/bash
sudo /etc/init.d/openvpn start_

_sudo nano disableVPNClient.sh_
copy the content below and save the file
_#!/bin/bash
sudo /etc/init.d/openvpn stop_

**Adjust the files permisssion**
sudo chown www-data enableVPNClient.sh
sudo chgrp www-data enableVPNClient.sh
sudo chown www-data disbaleVPNClient.sh
sudo chgrp www-data disableVPNClient.sh
sudo chmod +x enableVPNClient.sh
sudo chmod +x disableVPNClient.sh

**Addition of the 2 scripts for sudo execution exception in sudoers**
open the file for edition with the following command
_sudo visudo_
add at the end the 2 lines below and save the file
_www-data ALL=(ALL) NOPASSWD:/etc/raspap/hostapd/enableVPNClient.sh
www-data ALL=(ALL) NOPASSWD:/etc/raspap/hostapd/disableVPNClient.sh_

**Modification of functions.php file**

replace the beginning of the function up to the "// parse client settings" line by

function DisplayOpenVPNConfig()
{
    exec('cat '. RASPI_OPENVPN_CLIENT_CONFIG, $returnClient);
    exec('cat '. RASPI_OPENVPN_SERVER_CONFIG, $returnServer);
    exec('pidof openvpn | wc -l', $openvpnstatus); // Get the Public IP Important for VPN
    exec('wget  http://ipinfo.io/ip  -qO -', $public_ip);

    if ($openvpnstatus[0] == 0) {
        $status = '<div class="alert alert-warning alert-dismissable">OpenVPN is not running ['.$public_ip[0].']
                                        <button type="button" class="close" data-dismiss="alert" aria-hidden="true">×</button></div>';
    } else {
        $status = '<div class="alert alert-success alert-dismissable">OpenVPN is running ['.$public_ip[0].']
                                        <button type="button" class="close" data-dismiss="alert" aria-hidden="true">×</button></div>';
    }

**still in the same file, in the function SaveTORAndVPNConfig() adapt the changes as below**

    _} elseif (isset($_POST['StartOpenVPN'])) {
        echo "Attempting to start openvpn... ";
        //exec('/usr/bin/sudo /etc/init.d/openvpn start', $return);
        exec('sudo /etc/raspap/hostapd/enableVPNClient.sh', $return);
        foreach ($return as $line) {
            echo htmlspecialchars($line, ENT_QUOTES).'<br />' , PHP_EOL;
        }
    } elseif (isset($_POST['StopOpenVPN'])) {
        echo "Attempting to stop openvpn... ";
        //exec('/usr/bin/sudo /etc/init.d/openvpn stop', $return);
        exec('sudo /etc/raspap/hostapd/disableVPNClient.sh', $return);
        foreach ($return as $line) {
            echo htmlspecialchars($line, ENT_QUOTES).'<br />' , PHP_EOL;
        }_

**finally still in the same file, correct the condition** ( Ligne 231 in the original code) as i guess it's a bug, replace
if ($hostapdstatus[0] == 0) {
by
_if ($openvpnstatus[0] == 0) {_

-r45635

feh123, 2019-08-02:
I have an ovpn file and a vpn account. However Raspap still hangs at adding the ovpn file. (Currently I can only use v1.5 - 1.5.1 does not work on my RPI 3B running Buster Lite).
It would be great to get this working! As far as I can tell there are very few ways to have an access point with a VPN on a raspberry  pi.

billz, 2019-11-09:
@r45635 much appreciated. OpenVPN management via the UI is on the near-term roadmap

billz, 2019-11-15:
Will be rolling out an admin interface for managing OpenVPN client configuration shortly. Beta testers welcome.

r45635, 2019-11-16:
@billz I could sepend some time on the beta when ready. I've a raspberry for such purpose.

feh123, 2019-11-16:
Happy to test, I have RPI 3+B currently running RaspAP I could use.

rohitg76, 2019-11-16:
Great development! Count me in.

Regards,
Rohit

billz, 2019-11-16:
Great, I appreciate the help. This feature is closer to alpha than beta status 😉 so you will need to do a few things first.

1. (re)install RaspAP. this is important as an OpenVPN auth control script has been added along with updates to sudoers.
2. install openvpn, if not already present:
```
sudo apt update
sudo apt install openvpn
```
The openvpn version I have on Buster is 2.4.7-1.

3. Add `iptables` rules to forward traffic from the OpenVPN tunnel `tun0` to `wlan0` (substitute wlan0 for an alternate interface, if needed)...
```
sudo iptables -t nat -A POSTROUTING -o tun0 -j MASQUERADE
sudo iptables -A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i wlan0 -o tun0 -j ACCEPT
``` 
4. enable the OpenVPN management UI by changing this setting to 'true' in `includes/config.php`:
```
define('RASPI_OPENVPN_ENABLED', true);
```
5. go to **Configure OpenVPN** in RaspAP, enter your login details and upload a valid .ovpn file.  
 
[image]

6. Stop/start the openvpn service.
7. Configure hostapd as normal and start the AP.
8. Connect a client to the AP and visit https://ipinfo.io/ip or http://geoiplookup.net/

A few notes: client logging isn't functional yet. It would be nice to report the public IP in the UI when connected, will add this shortly. Check for errors with `sudo systemctl status openvpn-client@client` and `journalctl --identifier openvpn`. 

Report any bugs or other issues here. Thanks :)

feh123, 2019-11-16:
I have just followed your instructions after doing a fresh install onto Buster Lite (RPI 3+B).
I uploaded my VPN provider openvpn file, and added my id and password. Initially I think in error I saved these settings before hitting start vpn. This took me into a loop and I rebooted to escape. On my second go I just completed the form and hit start. This worked and an ip check correctly showed my VPN server address. Currently all my usual devices have connected to this AP without a problem.
Looks good!
Can I try to get this to work with a wlan1, wlan0 set up?

billz, 2019-11-16:
> Initially I think in error I saved these settings before hitting start vpn

That may be a UX issue, actually. Choosing 'save settings' handles upload of the .ovpn file and auth credentials. Maybe it's useful to indicate an unconfigured state before starting the service 🤔 

> This took me into a loop and I rebooted to escape

Yikes, can you elaborate on this a bit?

> Currently all my usual devices have connected to this AP without a problem

Excellent 👍 

> Can I try to get this to work with a wlan1, wlan0 set up?

Yes, with different iptables rules. I'd like to focus on the basic wlan0 configuration first.

Thanks for the feedback.

feh123, 2019-11-16:
I have closed and re-opened Raspap several times and openvpn always works yet I have never used the save settings button. By loop I meant that the openvpn up icon on the top RHS continually flashed red and nothing else happened. I thought I might get an error message but the software seemed to have stalled. Maybe I did not leave it long enough? Even though it seemed stalled the AP wlan0 was active.
I should have said for completeness that I installed the Raspap via the quick installer accepting all the recommended settings. 
Really nice to have it working!

billz, 2019-11-16:
@feh123 ok I see what you mean. I pushed an [update](https://github.com/billz/raspap-webgui/commit/e0bb18b283641c14604f53d2d39b1cbae1ce5b95) to the Quick Installer that fixes this. you can reinstall or edit sudoers manually with this change.

fustinoni-net, 2019-11-16:
Hi,
not sure if at this point this can be of any help, but you can have a look at:

https://github.com/fustinoni-net/raspap-webgui/blob/dev/includes/openVPN_manager.php

It use an unix socket connection to get the openvpn staus in real time.

It works for my poupose very well. I should have shere before (like the wifi management part)

Your is a really great work!!! Thanks a lot 

[image]

[image]

billz, 2019-11-17:
@fustinoni-net will give it a look, thanks

feh123, 2019-11-17:
Just to say I tried the installation on a RPI4 - I have installed the latest Raspap onto Buster Lite using the same procedure as on my 3+B.
However on save settings now I get the messages successfully uploaded file and authority. So all looks good however on start openvpn the system seems to get stuck and open vpn is not initiated. Is there something about the RPI4 or have I made a mistake something and should retry to whole installation again?

r45635, 2019-11-17:
I've tried on a fresh Raspberry PI3b+ with the "Raspbian Buster Lite"
distribution  and OpenVPN (2.4.7 arm-unknown-linux-gnueabihf)

I met 2 problems up to now:
my vpn configuration file has no password but the interface required one.
(i've understood the script is creating an auth file and adding a line at
the end of the ovpn configuration file but i don't need as i have a key
inside the conf)

I have uploaded a ovpn conf file. The upload is fine and place the
client.conf file in /etc/openvpn folder.
Unfortunately that does not work because the service : "
openvpn-client@.service"
expect to have the configuration file in /etc/openvpn/client/ directory

look my file:
sudo cat /lib/systemd/system/openvpn-client@.service
[Unit]
Description=OpenVPN tunnel for %I
After=network-online.target
Wants=network-online.target
Documentation=man:openvpn(8)
Documentation=https://community.openvpn.net/openvpn/wiki/Openvpn24ManPage
Documentation=https://community.openvpn.net/openvpn/wiki/HOWTO

[Service]
Type=notify
PrivateTmp=true
*WorkingDirectory=/etc/openvpn/client*
ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind --config %i.conf
CapabilityBoundingSet=CAP_IPC_LOCK CAP_NET_ADMIN CAP_NET_RAW CAP_SETGID
CAP_SETUID CAP_SYS_CHROOT CAP_DAC_OVERRIDE
LimitNPROC=10
DeviceAllow=/dev/null rw
DeviceAllow=/dev/net/tun rw
ProtectSystem=true
ProtectHome=true
KillMode=process

I did a simple copy of the file for now to the correct directory and now
it's working (both the cmd "sudo systemctl status openvpn-client@client"
the interface) , i still have to go further.

r45635

fustinoni-net, 2019-11-17:
In may experience it's more convenient using the openvpn client not like a service, but started like an executable:

`define('OPENVPN_UNIX_SOCKET', '/tmp/openVpn');`
...
...
...
`exec( 'sudo openvpn  --config '.$fileName.' --management '.OPENVPN_UNIX_SOCKET.' unix  > /dev/null 2>&1 &');`

In this way you can specify all the necessary parameters you need, like the unix socket were you will get the opnevpn output.
You can also check if it's running:

```
    $openvpnstatus = null;    
    exec( 'pidof openvpn | wc -l', $openvpnstatus);

    if( $openvpnstatus[0] == 0 ) {
        $status = $status.'<div class="alert alert-warning alert-dismissable">OpenVPN is not running
                            <button type="button" class="close" data-dismiss="alert" aria-hidden="true">×</button></div>';
    } else {
        $isVPNRunning = true;
        $status = '<div class="alert alert-success alert-dismissable">OpenVPN is running
                            <button type="button" class="close" data-dismiss="alert" aria-hidden="true">×</button></div>';
    }

```
while the output socket data will tell you if the vpn is connected and if not why.

Very simple and it works! :-)

feh123, 2019-11-17:
Just an update on the RPI4 - it is working now - so something possibly to do with my VPN provider not connecting because I have done nothing but re-start the rpi after a couple of hours.
So both my 3+ B and 4 are working with just with the instructions from @billz .

billz, 2019-11-17:
@r45635 you're right, I looked more closely at the OpenVPN docs and found this. thanks also for pointing out the key auth, I neglected that test case. pushing updates shortly

billz, 2019-11-26:
Closing this as OpenVPN support is now integrated into the Quick Installer.
Thanks everyone for feedback & testing.

closed 2019-11-26 by billz