RaspAP/raspap-webgui #149: Worked fine, but I ruined after chaning "Configure Networking" and "Configure DHCP Server"

zeus-dx, 2018-01-20:
Hello.
So I ruined it.
It was working fine with Internet Access and DHCP Server and then I played with the settings.
I wanted to disable DHCP because I don't like having two DHCP servers. 
Every device I use, I want it to be connected to my default gateway: 192.168.0.1
For convenient IP accessing.
So I did the following changes:

- Configure Networking > wlan0 > static ip
`Static IP Options
IP Address: 192.168.0.254
Subnet Mask: 255.255.255.0
Default Gateway: 192.168.0.1
DNS Server: 192.168.0.1
Alternative DNS Server: 8.8.8.4`

- Configure DHCP Server
`DHCP server settings
Interface: wlan0
Starting IP Address: 192.168.0.200
Ending IP Address: 192.168.0.220`

Unfortunately the settings did not work
So I tried restoring them to their default values.
I deleted everything I added in  Configure Networking (empty cells)
Restored DHCP to its default settings.

Now my other devices won't get an IP address or they will get one but cannot access Internet neither they can access the default gateway 192.168.0.1 (if i use the settings I mentioned

How do I make RaspAp work again?
Here are some screenshots:

https://s26.postimg.org/fufm1tweh/firefox_2018-01-20_13-55-49.png
https://s26.postimg.org/7c65xhxll/firefox_2018-01-20_13-56-03.png
https://s26.postimg.org/7oxk3onl5/firefox_2018-01-20_13-56-15.png

 Raspberry Pi hardware : Pi 3 Model B
* Raspbian version : Stretch
* RaspAP Quick Install

billz, 2018-01-20:
@zeus-dx RaspAP includes an uninstaller that will restore your original settings from a backup created when you ran the Quick Installer. From the project folder, execute `installers/uninstall.sh`. Alternatively, you can also make these changes manually by copying files from the `/backups` folder if you wish.

Hope this resolves your issue.

bderry71, 2018-01-27:
Sorry to post a me too, but I did the same thing when was trying to integrate Tor into the config. When looking for this project I found another one that has the same name but they implemented it in pythong instead of php. Haven't tried theirs yet as their installation process seemed less refined than yours. 

But in relation to what @zeus-dx  was saying, I think he was looking for a reset to factory button in your web-gui so that he could go back to a known working condition. Like what you get with any WiFi Router. I would too. Also your Network config page is not showing current config correctly.

Thanks for what work you have done so far. I'm using it on a RPI2 with Edimax adapter. I would like to be able to do all funtionality from the webui and that includes Tor/VPN setup.

jensjakob, 2018-04-13:
I think this is related to Issue https://github.com/billz/raspap-webgui/issues/184 - take a look at `/etc/dhcpcd.conf` -mine looked wierd but I didn't grab a copy of it before I fixed it.

billz, 2019-03-04:
Refer to #184.

closed 2019-03-04 by billz