RaspAP/raspap-webgui #151: WiFi AP+Client, possible? Nor working for me

hallard, 2018-01-25:
### Wifi Client and AP same time
I tried to search but I didn't found clear answer, is it possible to do WiFi AP and Client same time?
(no wired, just wifi), may be just not possible?

### Your environment
* Raspberry Pi 3 Model B
* Raspbian version Stretch lite
* RaspAP Quick Install

### Steps to reproduce
Fresh install, setup my PI with correct `wpa_supplicant.conf` and reboot, then Pi connected to my LAN and internet. Then quick install RaspAP. I setup RaspAP AP channel same as client just in case but same thing. All works but no AP visible

### Expected behaviour
When it connect as client, I do net get AP working, if I deleted my client network if `wpa_supplicant.conf` and reboot then AP is working.

kikil592, 2018-02-18:
Same material, same problem for me: AP is not visible when using wifi at the same time. I've tried all wireless mode (a,b,g) and all channel (1-14) in configure hotspot but no result

GevaudanBeast, 2018-04-24:
Same problem for me with Pi Zero W and USB dungle

billz, 2018-05-24:
@netrosec we have included an uninstaller for your convenience `installers/uninstall.sh`

hallard, 2018-05-24:
@netrosec 
For zero I created this [procedure
](https://github.com/ch2i/LoraGW-Setup/blob/master/doc/AccessPoint.md) 
Works fine if you need it ;)

billz, 2018-05-28:
@hallard thanks for including this procedure. Indeed, there are extra steps for the Pi Zero that are not provided by the quick installer or covered in our manual instructions

classofoutcasts, 2018-06-10:
It is possible to have an AP and wifi client running at the same time on the internal wifi adapter.
It has it's limitations regarding efficiency, but with only a few clients connected to the AP, it seems to work well.
Run [this](https://gist.github.com/lukicdarkoo/6b92d182d37d0a10400060d8344f86e4) script on a cleanly installed system, and perform a manually install of RaspAP afterwards. If problems occur, like network scanning which doesn't work as advertised you'll probably want to experiment with broadening the sudo permissions on the `wpa_cli` command.

This worked for me. Have it running fine on a Pi Zero W without a second USB wireless adapter.

SmithtekWA, 2018-07-13:
Hi Classofoutcasts. 
Would it be to much if i asked for a image of your Pi zero with a working RaspAP.
I have spent nearly a week trying to get this to work exactly how you describe. 

Thanks in advance.

Dan

flatsiedatsie, 2018-08-07:
This is the latest version of the script, and it seems to be working for most people:
https://github.com/lukicdarkoo/rpi-wifi

It would be great if RaspAp would allow people to connect to an existing network.
- user connects to the Pi acting as a captive hotspot
- user selects existing network's SSID and type in the password
- Optional: user selects if the networks should be bridged

billz, 2018-11-10:
I've added @hallard's procedure for the RPi Zero to our [wiki](https://github.com/billz/raspap-webgui/wiki/FAQs#pizero). Closing this for now, thanks all for the input.

closed 2018-11-10 by billz

Motyak, 2019-03-16:
Didn't work on my RPI3 with Internet access on wlan0, i executed the 'configure' script that I think stopped too early. I rebooted the pi like told by the script and then I got the error 'couldn't communicate with wpa_supplicant' and 'there was an error running option N2 Wi-fi'. Now i can't connect to any wifi.

billz, 2019-03-17:
@Motyak please refer to #300. This is currently in progress.