RaspAP/raspap-webgui #152: No information on Interface Statistics

hjacas, 2018-01-25:
### Subject of the issue
No Received or Transferred Packets/Bytes on Dashboard Interface Statistics

### Your environment
* Pi 3 Model B
* Raspbian Stretch (relese date 2017-11-29)
* RaspAP Quick Install
* No Other software or services running

### Steps to reproduce
* Install using the quick method
* Restart the pi
* Access through the web interface, change passwords and assign static IPs to the network interfaces (eth0 and wlan0)
* Access through ssh (from LAN) and add the firewall rules (with persistence mechanisms) and enable ip_forward
* Restart the pi again
* Connect via wifi: OK
* Access the Internet: OK
* But I do not have data traffic statistics

Note: 
Installation and Configuration no more than 15 min. Great!!!

### Expected behaviour
Interface Statistics session on Dashboard must be show information related to traffic.

### Actual behaviour
Interface Statistics

Received Packets No Data
Received Bytes No Data

Transferred Packets No Data
Transferred Bytes No Data

itshaadi, 2018-01-27:
i had the same issue. this could be a bug because the output of `ifconfig wlan0` is this

```bash
...
        RX packets 32793  bytes 5096198 (4.8 MiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 39938  bytes 36256917 (34.5 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

and the [regex for extracting this data](https://github.com/billz/raspap-webgui/blob/master/includes/dashboard.php#L30-L37) is incorrect [here is the proof](https://regex101.com/r/eZ2Fke/1) and [here](https://regex101.com/r/uJa2Oz/1/) is a way to correct this bug

billz, 2018-02-11:
@itshaadi thanks for doing the regex check. We transitioned to using `ip` rather than `ifconfig` in [this commit](https://github.com/billz/raspap-webgui/commit/47f6366bf9092f1bcf31af5cf0f12d5a9ce7e5dd#diff-a2159bbaf42b7f61ddba68fff6623118) and it appears we lost RX/TX stats in the process. @hjacas  this is indeed a bug that we need to fix.

heneraloysat, 2018-09-07:
@billz I tried adding this exec( 'ip -s link show ' . RASPI_WIFI_CLIENT_INTERFACE, $return ); under   exec( 'iwconfig ' . RASPI_WIFI_CLIENT_INTERFACE, $return );

this is the output for me
[image]

closed 2019-01-23 by billz