RaspAP/raspap-webgui #155: WPA2-Enterprise is there any functionality to setup client access to a WPA2-Enterprise access point?

danriches, 2018-02-08:
### Subject of the issue
Cannot see WPA2-Enterprise configuration parameters

### Your environment
* Raspberry Pi hardware Pi 3 Model B
* Raspbian version Jessie
* RaspAP Manual setup
* None, just Samba and eventually ReadOnly Filesystem

There doesn't seem to be a way of adding the .PEM certificate to the system to allow connection to a WPA2-Enterprise access point in client mode. Ie I want the RasPi 3 to connect to a WPA2-Enterprise access point, using wlan1 as we've disabled the onboard wlan0 because the device is inside a metal case.

Thanks and kind regards, Dan Riches

SirLagz, 2018-02-13:
No WPA2 Enterprise support yet, We'll add it to the todo list.

billz, 2018-10-14:
Added to the todo list. Closing this for now, thanks

closed 2018-10-14 by billz