RaspAP/raspap-webgui #156: Proposed enhancement - customisable configuration

njkeng, 2018-02-11:
### Subject of the issue
Customizable configuration of RaspAP

Hi Bill and Joseph and Lawrence

I'm keen to add customisation / configuration options and a couple of customisable pages to RaspAP.  These can be used by developers when deploying RaspAP in custom applications and in devices other than pure access points. The enhancements I am proposing will also specifically address issues #91 and #149, but that's not exactly why I'm making the changes.

First, some background:

I am developing two personal projects at the moment.  The projects are quite different to each other, but I want the bulk of the configuration to use the same mechanism.  The projects I am building need to be configured:
- By non-technical users
- Using a nice-looking interface
- Headless and wireless
- Without a dedicated app, so web based
After configuration, the device needs an internet connection, so must remain connected to a local wifi as a client.

When the device is not yet configured by the user, it needs to operate as an access point.  The user can connect to the device's wifi access point and view the device's web pages.  The web pages allow re-configuration of the Pi as a client on a local wifi. After successful wifi connection, the access point is turned off and the device is visible as a web server on the local wifi. Additional configuration or re-configuration is done on the local wifi.  This is great for a non-technical user.  All configuration is done via a pretty interface and needs no special equipment.  Any wifi device that can display web pages will do.

I didn't know of any such system for RaspberryPi and I assumed I had to develop such a system myself.  I stumbled across your useful project while I was searching for code snippets for configuration of wifi networks. Hey presto, all of the wifi part of the project is done! The web interface looks great on both desktop and mobile devices. 

Now to the mods I want to make to RaspAP.
For my projects I only need the wifi client functionality from the standard RaspAP.  I need to turn all the other RaspAP pages off, so I want to make all RaspAP pages "optional".  I also need to add new configuration pages for which I will develop my own /include/?.php scripts.

I believe the changes I wish to make will have use for many projects and applications.  That is why I want to contribute my modifications to the RaspAP project.  I'm going to be making these modifications for my purposes anyway, so it's no problem for me to contribute them back to the project.  As examples of where these additional options are extremely useful, the following describes the two projects I am working on.

Tide clock.
I am a keen kayak paddler and an occasional fisherman, so it's important to me to know what the tides are doing.  I am developing a tide "clock" that will use RGB LEDS to display the current tide level, whether it's rising or falling, the current temperature and the current wind speed.  I'm going to attempt to display all this on 30 RGB LEDs with some calming, unobtrusive animation, hopefully reminding the viewer of ocean waves. In order to get all of this data, the clock needs to read live values from a web page. The tide clock configuration interface will allow the user to select a web page with data for their location, anywhere on earth. The configuration will also let the user set max and min values for temperature and wind speed.  5degC may be a reasonable temperature for fishing for some people, but it's too cold for me!

Bike helmet camera.
I do a lot of cycling and I'd like to carry a forward-facing camera on my bike.  None of the commercial products suit my needs, so I'm going to build my own using a Pi Zero W.  The configuration interface for this device will allow the user to adjust settings such as video quality, auto shutdown timers and the like.  I also want to implement auto uploading of video to services such as YouTube, Dropbox or in my case, my local NAS. The user will be able to set up these services and test the configuration directly from the configuration page. Whenever the device is within wifi range, all video data will be automatically uploaded to the preferred destination.

So as you can see, it would be great if I could hand non-technical people either of these devices in a default state that they could configure the device themselves using only a web browser.  

Other notes:
With whatever changes I make, I will ensure that the RaspAP interface will act exactly as it does now unless it is specifically reconfigured by the developer. The interface appears to be mature and has all of the necessary features for a flexible AP. I won't change the functionality of the default interface.  I wish only to add options if a developer chooses to use them.

On to the specific changes to RaspAP.

Part 1: Add RaspAP custom configuration options

Mods to index.php
Modify index.php to enable/disable ANY side menu items as per flags set in config.php.  This has already been implemented for TOR relay and OpenVPN, so I will copy this format to look for the new flags.
Add two new items to the side menu for custom configuration. For usability, add one near the top of the functions list and one near the bottom. Both the custom menu item names and icons will be defined in /includes/config.php
Option to replace the RaspAP banner image with a developer-defined one.  If this option is used, RaspAP will display the user image PLUS the existing RaspAP image.  The text will be changed to "Powered by RaspAP" instead of "RaspAP".  I'll have to figure out a more suitable class for this text.  It's too big to mix with a second image.

New php scripts in the /includes directory
Add two demo/placeholder functions in the /includes directory, one for each new custom page.
For both of my own projects, I need to have quite a few configuration inputs.  So, since I need to figure this out anyway, so instead of having blank pages, I may as well do some well-commented general cases as examples. You know, here's how to add a button and a text input field.  Here's how to make tabbed sections.  Here's how to read options data in from a file and write updated data back out to a file.

Mods to /includes/config.php
Add boolean flags to enable / disable of ALL side menu items.  Ensure default flag values show the menu items as per current Raspap system.
Add two new placeholder menu items for custom configuration scripts. Add variables so that the developer can assign page names. Perhaps default these to "Settings" and "Advanced".
Add variables so that the developer can assign fa- icons from font-awesome.css.  Perhaps default these to "fa-gear" and "fa-star".
Enable / disable physical reset button. (See issue 2 below)
Enable / disable custom banner image or default RaspAP image.
IMPORTANT: All values will default to the configuration as currently displayed in RaspAP. Anything else will be used only if explicitly enabled by the developer.

Mods to readme.md
Update readme.md to indicate that lots of display and customization options are contained in the /includes/config.php file. Explain the new functionality of the flag options and the two custom configuration php menu items.

Once again, default values for all of this will make RaspAP look and behave exactly as it does now.

Part 2: Physical reset button

I think this is an extremely useful addition to the RaspAP system regardless of the application.  In the event that the configuration gets stuffed up or if (for example) the SSID of the local wifi changes, the physical reset button will restore all configuration back to default. There will be no need to resort to the terminal to sort things out. This will specifically address issue #149

Mods to common.sh
Currently, the default config files are MOVED from $webroot_dir/config to their destination folders.  I will change default_configuration() function to COPY these files instead.  Then, when a reset is performed, the original configuration files still reside in the  $webroot_dir/config directory.
While I'm modifying common.sh I will correct a few typos where "raspian" is used instead of "raspbian"

[... 1 lines cut ...]
Install a PHP dependancy manager called Composer. https://getcomposer.org/doc/faqs/how-to-install-composer-programmatically.md
Use Composer to install PiPHP/GPIO library https://github.com/PiPHP/GPIO
Use a PHP script to monitor a GPIO pin for reset. The script will set up the button as an interrupt.
[... cut ...]
Now, because these changes are inter-linked I think it best if I create a single branch and perform all of these changes at once.  Please let me know if you think there's a better way to attack this.

So, what do you think?  This is what I need for my projects, but it's a significant change in philosophy to RaspAP as it stands.  I have no idea whether you'll think this change in philosophy is good or bad. Hopefully you can see I've put a fair bit of thought into this already, so I'm comfortable that all of this will work well. Are you interested in me contributing this to the RaspAP project?

Cheers,

Nathan.

njkeng, 2018-02-11:
Dammit!  Looks like emmanuelgeoffrey beat me by about 4 hours for some of these changes.  Let me know what you think about the rest.

njkeng, 2018-02-11:
Sorry about the closes and opens.  I'm still learning how to drive this thing.

PRO2XY, 2018-11-11:
Have you since been able to make any of the mods? If so, you could share the code to be reviewed here.

njkeng, 2018-11-12:
Yes, I have committed all changes to the pull request Customisable configuration #167. Nine months ago.

PRO2XY, 2018-11-20:
Great! I had missed that :-)

billz, 2019-03-04:
Thanks for the PR @njkeng. Your many contributions have proven invaluable to the project. Finally making time to circle back to this issue to give it the attention it deserves.

As you noted, @emmanuelgeoffray added some of this functionality already and more changes have been made along these lines since then. At this stage, I think we've struck a good balance between ease of customization (eg., all principal configuration UI's are easily enabled/disabled) and a fully tweakable project with custom strings, sample code and template files. Judging by the many forks of RaspAP, developers appear to have no trouble customizing the project to suit their needs. Some forks are simply a rebrand, which I of course have no issue with. If someone feels a sense of ownership by personalizing it, so much the better. Such is the nature of open source.

Re: the factory reset, either a physical button, "soft" UI option or both, this is an excellent idea and the work you've done in PR #172 is outstanding. Unfortunately, around the same time we released [multilingual support](https://github.com/billz/raspap-webgui/releases/tag/1.4.0)  for RaspAP with seven new languages. The factory reset button PR adds a lot of new strings, all of which would need to be localized. Several more translations have been added and I have a even more waiting in the wings. Hence the long delay in merging it. 

Perhaps we can refactor PR #172 as an official add-on to the project with a separate repo. Thoughts?

njkeng, 2019-03-07:
@billz Yes, considering all of the other changes that have been going on, a separate to for an ad-on sounds like the best option. Please let me know how it is best to proceed from here.

billz, 2020-03-13:
@njkeng circling back to this, I'd like to cherry pick #172 for the python and shell portions, commit them to the project and add your documentation to the wiki, crediting you as the author. This way users can integrate the functionality on their own. The UI has diverged pretty far since then, so I think it's best to retire this piece for now. How does this sound?

njkeng, 2020-03-13:
No problem Bill. Whatever is useful to you.

Nathan

Sent from BlueMail<http://www.bluemail.me/r?b=15824>