RaspAP/raspap-webgui #157: The webgui works only once

Jalex-h, 2018-02-12:
### Subject of the issue
First of all, I want to thank for all the good work.

On a fresh Stretch install, I first installed hostapd and dnsmasq. Then I installed the webgui with the quick install. After the reboot, everything works fine. But when I reboot again, the router didn't start up.  There's no wifi-signal.
So I start all over again. Did the fresh Stretch install and then installed hostapd, dnsmasq and finally the webgui with the quick install. And got the same result. Everything works fine until I reboot. 

When I do 'lp agx', I see some lines that indicates that dnsmasq, lighttpd, php and dhcp are running but there is nothing indicating hostapd is running:
  401 ?        Ss     0:00 /sbin/dhcpcd -q -b
  429 ?        Ss     0:00 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplican
  484 ?        Ss     0:00 /usr/bin/vncserver-x11-serviced -fg
  494 ?        Ss     0:00 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplican
  507 ?        Ssl    0:00 /usr/sbin/lightdm
  559 ?        S      0:01 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsm
  617 tty7     Rsl+   1:00 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/ligh
  621 ?        Ss     0:00 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
  651 ?        Ss     0:00 /usr/bin/php-cgi
  690 ?        S      0:00 /usr/bin/php-cgi
  691 ?        S      0:00 /usr/bin/php-cgi
  692 ?        S      0:00 /usr/bin/php-cgi
  693 ?        S      0:00 /usr/bin/php-cgi
(full dump :[allex.txt](https://github.com/billz/raspap-webgui/files/1717888/allex.txt))

In /etc/rc0.d (and rc1.d till rcS.d) there is a K01hotapd-file. In this file there is a reference PIDFILE=/run/hostapd.pid. I tried to execute this file with Run, but I got the message that the file doesn't exist.

### Your environment
* Raspberry Pi hardware: Pi 3 Model B with an Alfa Tube-N acting as wlan1 (the wifi-client)
* Raspbian version: Stretch
* RaspAP Quick Install
* Other software or services running with RaspAP? VNC Server

### Steps to reproduce
Install hostapd, dnsmasq and finally the webgui with the quick install, ending with a reboot.
When the Raspberry is up and running again, reboot again.

### Expected behaviour
The router should start, including the wifi-network

### Actual behaviour
No wifi-network

njkeng, 2018-02-14:
Jalex-h,
The quick installer installs both dnsmasq and hostapd.  See if it works if you just use the quick installer on a fresh Stretch install.

Jalex-h, 2018-02-14:
Thanks for your reply. But unfortunately that isn't the solution.
I start all over again and only installed dnsmasq and hostapd and configured them. The result is that now nothing happens.
So it looks like the problem is not in raspap but somewhere else. But I haven't got a clue where...

closed 2018-02-14 by Jalex-h