RaspAP/raspap-webgui #158: Open RaspAP web gui when device connects to hotspot

e-nathan, 2018-02-14:
### Subject of the issue
Hello, Is it possible to open the RaspAP WebGUI page on browser immediately when a user connects to the hotspot (similar to a captive portal) ? 

Help much appreciated, thanks.

### Your environment
* Raspberry Pi hardware :Pi 3 Model B
* Raspbian version : Stretch
* RaspAP: Quick Install

kellymonster25, 2019-03-02:
I have the same question, but im using the app nodogsplash from GitHub also, but my problem is that my nodogsplash needs a gateway address, and i put 10.3.141.1, the address of the raspberry pi; but whenever i connect a new device, it never works. Do i need to tell raspap-webgui that it has a captive portal?

billz, 2019-03-02:
The nodogsplash project works just fine with RaspAP and is recommended over other methods. No configuration changes are needed with raspap-webgui, however you will need to modify some default settings in the nodogsplash config. 

The nodogsplash `GatewayInterface` should be set to the interface RaspAP runs on (`wlan0` is used by the Quick Installer). You will also need to change the `GateWayAddress` to `10.3.141.1`. 

` sudo vim /etc/nodogsplash/nodogsplash.conf`

```# Parameter: GatewayInterface
# Default: NONE
#
# GatewayInterface is not autodetected, has no default, and must be set here.
# Set GatewayInterface to the interface on your router
# that is to be managed by Nodogsplash.
# Typically br-lan for the wired and wireless lan.
#
GatewayInterface wlan0
```

```
# Parameter: GatewayAddress
# Default: Discovered from GatewayInterface
#
# This should be autodetected on an OpenWRT system, but if not:
# Set GatewayAddress to the IP address of the router on
# the GatewayInterface.  This is the address that the Nodogsplash
# server listens on.
#
GatewayAddress 10.3.141.1
```

Give the nodogsplash service a kick:

`sudo systemctl restart nodogsplash`

...and verify that the service is running:

`sudo systemctl status nodogsplash`

```● nodogsplash.service - NoDogSplash Captive Portal
   Loaded: loaded (/lib/systemd/system/nodogsplash.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-03-02 07:55:11 UTC; 11min ago
  Process: 3404 ExecStart=/usr/bin/nodogsplash -d 5 $OPTIONS (code=exited, status=0/SUCCESS)
 Main PID: 3405 (nodogsplash)
   CGroup: /system.slice/nodogsplash.service
           └─3405 /usr/bin/nodogsplash -d 5
```

Point your browser to http://10.3.141.1 and you should see the default nodogsplash page. 

[image]

kellymonster25, 2019-03-02:
well, i did edit the nodogsplash.conf, and it is running because when i go to 10.3.141.1, it shows the splash page, but when i try to see the status, it says that it cant find the nodogsplash.service
[image]

and it still wont show up on any other webpage i go to, my devices say theres no internet though the raspAP anyways

billz, 2019-03-03:
@kellymonster25 I've described a method to install this on a typical RaspAP setup with lighttpd, provided by the Quick Installer. Given that your setup is atypical, I not really able to help you troubleshoot in this case. The best advice I can offer is to start with a fresh install of the latest Raspbian, run the RaspAP Quick Installer, follow the instructions for installing nodogsplash on Debian and apply the configuration settings I've provided. 

Closing this issue for now. Feel free to reopen if needed, thanks.

closed 2019-03-03 by billz