RaspAP/raspap-webgui #163: RPi3B - No visibile client networks

ScottSturdivant, 2018-02-20:
### Subject of the issue
When I go to 'Configure WiFi Client', the list of networks is empty.

### Your environment
* Raspberry Pi 3 B v1.2
* 2017-11-29 Stretch Lite
* RaspAP Quick Install
* Other software or services running with RaspAP:  SSH

### Steps to reproduce

1. Install the latest stretch lite image onto a raspberry pi 3 B model.
1. Enable and start ssh:  `systemctl enable ssh && systemctl start ssh`
1. Execute the raspap-webgui quick installer:  `wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap`
1. Reboot.
1. Connect to raspap-gui, navigate to 10.3.141.1, then click on 'Configure WiFi Client'.

### Expected behaviour
I would expect to see several networks.  Instead, there are none.

### Actual behaviour
No networks are visibile.

Not sure how this is all supposed to work, but `wpa_cli scan && sleep 3 && wpa_cli scan_results` selects the `p2p-dev-wlan0` device and no results are displayed:

```
root@raspberrypi:/home/pi# wpa_cli scan && sleep 3 && wpa_cli scan_results
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
bssid / frequency / signal level / flags / ssid
```

If I manually specify `wlan0`, there are results:

```
root@raspberrypi:/home/pi# wpa_cli -i wlan0 scan && sleep 3 && wpa_cli -i wlan0 scan_results
OK
bssid / frequency / signal level / flags / ssid
de:9f:db:f3:27:02       2412    -41     [WPA2-PSK-CCMP][ESS]    tmp
40:b0:34:83:45:a6       2437    -48     [WPA2-PSK-CCMP][WPS][ESS][P2P]  DIRECT-A5-HP OfficeJet Pro 6970
6c:b0:ce:07:77:0f       2422    -87     [WPA2-PSK-CCMP][WPS][ESS]       Ginger
84:1b:5e:34:7f:14       2437    -90     [WPA2-PSK-CCMP][WPS][ESS]       NETGEAR35
cc:03:fa:de:ca:42       2437    -94     [WPA-PSK-CCMP+TKIP][WPA2-PSK-CCMP+TKIP][WPS][ESS]       HOME-CA42
00:03:93:e9:4e:a6       2412    -96     [WPA2-PSK-CCMP][ESS]    Kelly House
ee:9f:db:f3:27:02       2412    -41     [ESS]   bemyguest
```

This is my first time trying to use raspap-webgui so I'm not entirely sure that this should work, but I feel like it should.  No doubt I'm missing something simple...Thanks for any help.

diveyez, 2018-03-18:
Now with the recent updates, this wont even see HOSTAPD so +1 for getting it working.
I am going back to using create_ap

pagaille, 2018-03-24:
Any info or workaround regarding this issue ? Txs :-)

billz, 2018-03-24:
@ScottSturdivant thanks for the detailed debug info. I don't have an RPi3B handy (just ordered one) so can't repro this issue at the moment. That you were able to get output in the shell by manually specifying the interface is clearly a good indicator.

In order for this to work in the web UI, you'll need to modify permissions for the www-data user with `visudo` like so:

```
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan0 scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan0 scan
``` 

and refresh the configure client page. If the same output doesn't appear in the web UI, try:

`sudo journalctl | grep 'COMMAND=/sbin/wpa_cli'`

and look for any errors like "www-data : command not allowed".

Perhaps some other RPi3B users can contribute their findings. If wpa_cli requires the interface to be specified on this hardware, the quick installer will need to be updated.

pagaille, 2018-03-25:
Thanks @ScottSturdivant & @billz ! Just tested, it works correctly.

I could submit a PR but wouldn't this create a regression on other pit than the pi3b ?

billz, 2018-03-28:
@pagaille specifying the interface with `wpa_cli -i wlan0` works with the RPi1 & 2, although can't say the same for the Pi Zero. So this shouldn't create regression issues with those models. However, for this to be complete I think the Quick Installer should be updated to ask for the wireless adapter during setup, defaulting to wlan0. This value would then be written to sudoers. 

@ScottSturdivant my RPi3 arrived this morning and I reproduced this issue exactly as you described. A fix will be forthcoming shortly. Thanks

billz, 2018-03-28:
Thanks @ScottSturdivant and @pagaille for your valuable input.

closed 2018-03-28 by billz

pagaille, 2018-03-28:
@billz You're welcome. I'll test on the zero W one of these days.

commit 2a05284b (2018-03-28): Specifies interface with -i option and wpa_cli. Resolves #163