RaspAP/raspap-webgui #170: Pi3 with build in WiFi as RaspAP + USB dongle WiFI connected to other WiFi(box, smartphone 3G)

Guttata, 2018-03-05:
### Subject of the issue
Hi,

First of all, thanks a lot for this awesome utility.
I use it on a Pi which serves as a microcontroller for an autonomous rover car (to control motors, etc...)

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi 2 Model B, Pi 1 Model B+, Pi Zero)   ==> Pi 3
* Raspbian version (examples: Stretch, Jessie) ==> Stretch
* RaspAP Quick Install or Manual setup? ==> Quick install
* Other software or services running with RaspAP?

### Steps to reproduce
Plug an USB WiFi dongle

### Expected behaviour
In addition to the local AP, i would like to connect my Pi to internet over WiFi (using an USB dongle) (it's working using Eth0).
I don't want to have internet available for the client of the AP.

I just want to 
- Be able to connect to the Pi local AP (local communication with NodeJS hosted on the Pi),
- Connect the Pi to internet using USB WiFi dongle (to download new libraries, to send data to AWS, Azure)

Is it possible ?

### Actual behaviour
Once USB WiFi dongle inserted, i cannot select the new wifi interface to connect to my box Wifi (or smartphone wifi tethering)
If i restart the Pi, the AP is not working anymore.

diveyez, 2018-03-18:
in /etc/nerwotk/interfaces add 
auto <interface name>
allow-hotplug <interface name>

billz, 2019-04-13:
This setup is described in the [FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#interfaces).

closed 2019-04-13 by billz