RaspAP/raspap-webgui #173: Configure Networking is not working

arvjsr1802, 2018-03-13:
### Subject of the issue
Describe your issue here.

### Your environment
* Raspberry Pi hardware (examples: Pi 3 Model B, Pi 2 Model B, Pi 1 Model B+, Pi Zero)  
* Raspbian version (examples: Stretch, Jessie)
* RaspAP Quick Install or Manual setup?
* Other software or services running with RaspAP?

### Steps to reproduce
Tell us how to reproduce this issue.

### Expected behaviour
Tell us what should happen.

### Actual behaviour
Tell us what happens instead.

diveyez, 2018-03-18:
NONE of it is working.

billz, 2018-03-20:
@arvjsr1802 please re-submit this with the provided issue template. You can also enable logging via the UI with Configure Client > Advanced and copy-paste the logfile output in your issue. This will greatly assist with troubleshooting. thanks

closed 2018-03-20 by billz