RaspAP/raspap-webgui #174: Errors in 'common.sh' File

Austin-RR, 2018-03-18:
I am very new to this RaspAP and had some difficulty getting it to work with the "Quick Installer" method.  After reading some of the other posts on how to fix things, I did get RaspAP to work.  I looked a little further to try and identify what could have caused the problem in the first place.  There may be as many as 4 errors in 2 lines of code in the "default_configuration" function in the "installers/common.sh" file.  To be more precise, the errors are within the "lines" array and the subsequent for loop.

1. The first element in the "lines" array should have the forward slashes (/) escaped with backslashes (\\).
2. The sed command should be run with sudo or you get a "permission denied" error.
3. As the text "exit 0" may also appear in the comments in /etc/rc.local, the sed command should only look for a line that only contains "exit 0" by searching for: `^exit 0$`.
4. The sed command is incorrectly replacing "exit 0" with "exit0".

The original and then the corrected lines are as follows:

`'echo 1 > /proc/sys/net/ipv4/ip_forward #RASPAP'` (Original)
`'echo 1 > \/proc\/sys\/net\/ipv4\/ip_forward #RASPAP'` (Corrected)

`sed -i "s/exit 0/$line\nexit0/" /etc/rc.local` (Original)
`sudo sed -i "s/^exit 0$/$line\nexit 0/" /etc/rc.local` (Corrected)

On a clean Raspbian installation and making these edits prior to the RaspAP installation allowed RaspAP to install and operate fine.
______________
RPi 3 Model B
Raspbian Stretch (w/Desktop) 2018-03-13

diveyez, 2018-03-18:
Seems like this thing is broken, eh?

billz, 2018-03-19:
Pushed a fix for this. Many thanks @Austin-RR

commit 98ace523 (2018-03-19): bug fixes to default_configuration(). Resolves #174. Thanks @Austin-RR
closed 2018-03-19 by billz