RaspAP/raspap-webgui #178: [SSID NOT Visible] Pi2 Model B with TL-WN725N, AP is not accessible

Fisherzz, 2018-03-18:
### Subject of the issue
Pi2 Model B with Wifi dongle **TL-WN725N**, AP is not accessible

### Your environment
* Raspberry Pi hardware Pi 2 Model B
* Raspbian version Stretch (March 2018)
* RaspAP Quick Install
* Changed driver=rtl871xdrv (Since my WiFi Dongle is **TL-WN725N**) and channel=6 in /etc/hostapd/hostapd.conf

### Steps to reproduce
* DNSMASQ is RUNNING

> sudo service dnsmasq status 
> ● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
>    Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2018-03-18 17:56:14 UTC; 19min ago
>   Process: 433 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
>   Process: 390 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
>   Process: 369 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
>  Main PID: 427 (dnsmasq)
>    CGroup: /system.slice/dnsmasq.service

* HOSTAPD is RUNNING

> sudo service hostapd status 
> ● hostapd.service - LSB: Advanced IEEE 802.11 management daemon
>    Loaded: loaded (/etc/init.d/hostapd; generated; vendor preset: enabled)
>    Active: active (running) since Sun 2018-03-18 18:08:19 UTC; 8min ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 2047 ExecStop=/etc/init.d/hostapd stop (code=exited, status=0/SUCCESS)
>   Process: 2081 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/hostapd.service

### Actual behaviour
Cant find my SSID when I am looking from my mobile.

My requirement is I wish to use My PI as HotSpot which can share wireless internet with other device where in PI will get its  internet via Ethernet Cable.

Any help is appreciated.
Thank You!

billz, 2018-03-20:
@Fisherzz I only have experience with the Edimax Nano adapter and nl80211 driver. But there are some things you can do to troubleshoot this. Have you checked the syslog for error messages? Running hostapd from the command line can also give some clues as to what's happening. Check the output of `sudo hostapd -dd /etc/hostapd/hostapd.conf`

How did you install the rtl871xdrv driver? AFAIK it requires a patch from this repo https://github.com/pritambaral/hostapd-rtl871xdrv or a precompiled binary.

Fisherzz, 2018-03-20:
Thanks @billz for your response.

I will check with hostapd specific logs and reply, but as I said it was just running fine. Still will try to dig in deep.

For driver, I had referred [ a blog from 2013 via Google Search]( http://raspberry-at-home.com/hotspot-wifi-access-point/) where the author has mentioned below lines

> Because I had bloody RTL8188CUS chipset I had to replace original hostapd with the one prepared for my chipset.
> wget "http://raspberry-at-home.com/files/hostapd.gz"
> sudo gzip -d hostapd.gz
> sudo chmod 755 hostapd
> sudo cp hostapd /usr/sbin/

Although I will try the link you shared and will share logs of hostapd as well as status if it works out soon.

Again Thanks and I appreciate your time for responding.

Fisherzz, 2018-03-20:
Hi @billz,

Please find logs in attached file [Logs.txt](https://github.com/billz/raspap-webgui/files/1830567/Logs.txt) for `sudo hostapd -dd /etc/hostapd/hostapd.conf`

billz, 2019-03-12:
I can't determine the cause of the driver failing from the logs, sorry. 
I'd suggest either the Edimax Nano or maybe a newer RPi with an integrated wifi chipset. 

Closing this for now. Feel free to reopen if you have updates, thanks.

closed 2019-03-12 by billz

fustinoni-net, 2019-03-12:
Hi,
if your TL-WN725N is a version  2
IC:8853A-WN725N
FCC ID: TE7WN725N
and use the chip: rtl8188eus
you can try following the instruction at:
[https://www.raspberrypi.org/forums/viewtopic.php?t=62371](url)
for installing the right driver.

It works for me.