RaspAP/raspap-webgui #179: Is there any way I can contact you(billz) privately? I would really appreciate it.

doteric, 2018-03-20:
Hello.

I was trying search everywhere for a way to contact you in private, but I was unsuccessful in finding anyway. Maybe you could drop some email of yours so I can write there exactly what my question/issue is?
I would really appreciate it.

Thanks,
Eric

billz, 2018-03-21:
Our emails are in the phpDoc header of index.php 😉Feel free to get in touch

doteric, 2018-03-21:
@billz I looked through all your other social related accounts, website and projects, but totally didn't expect that email be in the index.php file, wow.

Anyway thanks and I'll email you as soon as possible with explanation.

You can close/delete this issue if possible.

closed 2018-05-24 by billz