RaspAP/raspap-webgui #180: RaspAP Install Error: Unable to update package list

Gourishmesta, 2018-03-21:
I am trying to run the command sudo wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
it gives me error while Rasp Install : Updating Source

> Ign:1 http://archive.raspberrypi.org/debian stretch InRelease                  
Err:2 http://archive.raspberrypi.org/debian stretch Release                    
  Connection failed [IP: 93.93.130.214 80]
Ign:3 http://mirrordirector.raspbian.org/raspbian stretch InRelease        
Err:4 http://mirrordirector.raspbian.org/raspbian stretch Release          
  Connection failed [IP: 93.93.128.193 80]
Reading package lists... Done
E: The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://mirrordirector.raspbian.org/raspbian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
RaspAP Install Error: Unable to update package list

### environment
* Raspberry Pi 3 Model B
* Raspbian version (Stretch)
* RaspAP Quick Install

billz, 2018-03-21:
Since wget for the Quick Installer succeeds, I'd rule out a DNS issue. Try `curl -v http://mirrordirector.raspbian.org/raspbian/dists/stretch/InRelease` and check `/etc/apt/sources.list`

billz, 2018-03-28:
Closing due to inactivity. Feel free to reopen if needed. Thanks

closed 2018-03-28 by billz

Kaushalop, 2019-07-10:
I ran into a very similar issue on the **Rpi Zero**
```
RaspAP Install: Updating sources
E: Command line option --allow-releaseinfo-change is not understood
RaspAP Install Error: Unable to update package list
```
Environment
```
RaspAP Install: Configure installation
Detected Raspbian 8.0 (Jessie)
Install directory: /etc/raspap
Lighttpd directory: /var/www/html
Complete installation with these va
```
ran the command `curl -v http://mirrordirector.raspbian.org/raspbian/dists/stretch/InRelease`

and this is what shows up in `/etc/apt/sources.list` 
```
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
```
is there something i am missing?

I have been following this article : https://howtoraspberrypi.com/create-a-wi-fi-hotspot-in-less-than-10-minutes-with-pi-raspberry/

@billz  Thanks in advance!

billz, 2019-07-10:
@Kaushalop refer to #343