RaspAP/raspap-webgui #181: RPi zero W Hotspot : SSID visible, password not accepted

harlock974, 2018-03-26:
### Subject of the issue
After a RaspAp install without error, the hotspot SSID is visible but the password is not accepted.
The Web GUI could be opened via an ethernet connection or a wifi connection manually configured.

### environment
* Raspberry Pi hardware : Pi Zero)  W
* Raspbian version : Stretch
* RaspAP Quick Install
* Other software or services running with RaspAP : no

When I stp/start the service, the log tell :

Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:1e:b8:d4 and ssid "raspi-webgui"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED

harlock974, 2018-03-26:
Further information : When the client attempt to connect, the web-gui log says :

wlan0: STA 68:17:29:88:22:1e IEEE 802.11: associated
wlan0: STA 68:17:29:88:22:1e IEEE 802.11: disassociated
wlan0: STA 68:17:29:88:22:1e IEEE 802.11: disassociated
wlan0: STA 68:17:29:88:22:1e IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 68:17:29:88:22:1e
wlan0: STA 68:17:29:88:22:1e RADIUS: starting accounting session 5AB87EE1-00000001
wlan0: STA 68:17:29:88:22:1e WPA: pairwise key handshake completed (WPA)
wlan0: STA 68:17:29:88:22:1e WPA: group key handshake completed (WPA)
wlan0: STA 68:17:29:88:22:1e IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 68:17:29:88:22:1e
wlan0: STA 68:17:29:88:22:1e IEEE 802.11: disassociated

billz, 2018-05-28:
@harlock974 thanks for the debug info. The Pi Zero needs some additional steps. I've added a link to the procedure in [our FAQ](https://github.com/billz/raspap-webgui/wiki/FAQs#ap-is-not-visible-or-cant-connect-to-ssid-with-the-pi-zero).

closed 2019-04-13 by billz