RaspAP/raspap-webgui #182: Changing http root directory "/var/www/html" to "/var/www/html.raspap"

Allroad, 2018-04-01:
### Subject of the issue
I will try to change :
1- the lighttpd port to 82 instead of 80
2- the path to /var/www/html.raspap instead of /var/www/html

1 => works great
2 => I got "error 403: forbidden" when connecting to web server

### Your environment
* Pi 3 Model B  
* Raspbian Stretch,
* RaspAP Quick Install
* JEEDOM ( apache/mysql/php7)

### Steps to reproduce
Install jeedom and after RaspAP
change lighttpd to port 82 and web root path on lighttpd to /var/www/html.raspap (with moving files to this path of course)

### Expected behaviour
I would like to have www/html with jeedom web pages and www/html.raspap with RaspAp

### Actual behaviour
I got on "error 403: forbidden" on raspap when i connect on port 82

billz, 2018-04-03:
Verify ownership for the `www-data` user/group on the html.raspap folder you created, and its contents. When in doubt, also check the lighttpd error log.

billz, 2018-04-16:
Closing due to inactivity. Feel free to reopen if needed, thanks

closed 2018-04-16 by billz

Oscar123456780, 2023-10-04:
Open again please! I have this issue but that www-data thing doesn’t work…