RaspAP/raspap-webgui #184: Things disappeared from dhcpcd.conf

jensjakob, 2018-04-13:
I was trying to get mobile broadband connection shared via both wlan0 and eth0 but suddenly `/etc/dhcpcd.conf` didn't include important settings about ip-address etc. This resulted in errors shown in `/var/log/syslog`. I think I was editing the dns for a few interfaces when this problem happened.

Problem is that the following lines was removed from `dhcpcd.conf`
```
interface wlan0
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=8.8.8.8 8.8.4.4
```

### My environment
* Raspberry Pi 3 Model B
* Raspbian Stretch 
* RaspAP Quick Install
* Docker (Home Assistant)

### Steps to reproduce
Haven't researched.

tourdownunder, 2018-10-21:
Thanks for this. I has this issue with a fresh install of raspbian today. The symptoms is that my wlan0 interface had the DHCP address and not the Static IP I assigned it.

billz, 2019-03-04:
RaspAP manages settings in `dhcpcd.conf` via the UI, so any manual changes will indeed be overwritten when choosing 'Save settings' from the DHCP menu. 

All relevant configuration files are backed up to `/etc/raspap/backups` when running the Quick Installer, however, so original settings can be restored from here if needed.

closed 2019-03-04 by billz