RaspAP/raspap-webgui #187: I can't find the options optional services code in the "Index.php"

gregk205, 2018-04-18:
I am trying to enable openvpn in the index.php but I can't find these lines anywhere. I am looking in the index.php file on your GitHub. Maybe I am looking in the wrong place. Thank you for this script. It works great so far. 

// Optional services, set to true to enable.
define('RASPI_OPENVPN_ENABLED', false );
define('RASPI_TORPROXY_ENABLED', false );

WhistleMaster, 2018-04-23:
[here](https://github.com/billz/raspap-webgui/blob/master/includes/config.php#L24)

billz, 2018-04-23:
@gregk205 the optional services seem to be creating confusion for many users. Per the readme, these are non-functional UIs and RaspAP does not come preconfigured with OpenVPN or Tor. There are several excellent projects that provide these services for the RPi, such as [pivpn](https://github.com/pivpn/pivpn). However, these are not supported by the project at this time.

gregk205, 2018-04-23:
Thank you @billz. I was able to enable openvpn thanks to @WhistleMaster help. I know that this isn't exactly what I was looking for but it's really a simple and beautiful setup you have here. I already have openvpn running on a server at home. I travel all the time for work so what I am trying to do is have my Pi Zero W act as my travel wireless access point and route all the traffic back thru my openvpn server at home. So far your script does half of that perfectly and with a simple and beautiful interface. I wish I knew more about scripting because I would love to help add to this project.   Thanks again for sharing this. It works perfectly for what you designed it for.

billz, 2018-06-06:
@gregk205 thanks for the kind words. This [webgui project](https://github.com/evilbunny2008/rock64-webgui) for the Pine64 has implemented an OpenVPN client interface. With some work it could be adapted to RaspAP.

[image]

billz, 2018-11-13:
Closing this for now. Tor and OpenVPN support are on the long-term project roadmap. Thanks

closed 2018-11-13 by billz