RaspAP/raspap-webgui #188: First APs in client scan

bluthen, 2018-04-22:
### The first access point is missing in client scan
When doing a client scan the first access point is missing for me.

### Your environment
* Pi 3 Model B
* stretch
* RaspAP Quick Install

### Steps to reproduce
Configure Wifi Client - Rescan

### Expected behaviour
Shows all access points

### Actual behaviour
The first one in the scan is missing.

wpa_cli v2.4

wpa_cli -i iface scan_results  Only had one header line not two.

In configure_client.php, changing
```
  for( $shift = 0; $shift < 2; $shift++ ) {
    array_shift($scan_return);
  }
```

to 

```
  for( $shift = 0; $shift < 1; $shift++ ) {
    array_shift($scan_return);
  }
```

ends up not removing the first access point result for me.

ScottSturdivant, 2018-05-04:
Just came to here report the very same issue, although on an RPi0W.  For reference, he's the output of my `scan_results` cmd, confirming that there's only a single header line:

```
bssid / frequency / signal level / flags / ssid
de:9f:db:f3:27:02       2412    -33     [WPA2-PSK-CCMP][ESS]    tmp
40:b0:34:83:45:a6       2437    -71     [WPA2-PSK-CCMP][WPS][ESS][P2P]  DIRECT-A5-HP OfficeJet Pro 6970
0c:51:01:e7:48:a8       2462    -87     [WPA2-PSK-CCMP][ESS]    RLK's Wi-Fi Network
ee:9f:db:f3:27:02       2412    -30     [ESS]   bemyguest
```

richardfat7, 2018-06-12:
same issue

commit ff438575 (2018-06-12): Fixes #188. Thanks @bluthen
closed 2018-06-12 by billz