RaspAP/raspap-webgui #190: AP not visibile if incorrect client password entered

ScottSturdivant, 2018-05-01:
### Subject of the issue
If I enter an incorrect password for the client network then reboot, the access point network is no longer visible.  I don't feel this is a problem with raspab-webgui persay, but I'd appreciate any help getting it solved or working around the issue!

### Your environment
* RPi0W + USB WiFi Dongle (Edimax N150: so I can do both AP + Client)
* Jessie, all up to date
* RaspAP Quick Install, then set AP to run on wlan1 (`/etc/hostapd/hostapd.conf`, `/etc/dhcpcd.conf`, `/etc/dnsmasq.conf`), this is because the client runs on wlan0.
* SSH is also running

### Steps to reproduce

1. Install raspap-webgui via the quick installer.
1. Modify `/etc/hostapd/hostapd.conf`, `/etc/dhcpcd.conf` and `/etc/dnsmasq.conf` changing the `wlan0` to `wlan1`.
1. Reboot
1. Scan networks, see raspap AP, connect, configure client WiFi.  Choose a network, enter the WRONG password.
1. Reboot

### Expected behaviour
I would not expect that the RPi0W was connected to the client network, but I would expect that the access point was still visible.  At that point I could connect to the AP, navigate to the webgui and correct the errant entry.

### Actual behaviour
The AP is not visible in the network scan.  By going into to `/etc/wpa_supplicant/wpa_supplicant.conf` and either removing the errant entry or correcting it to contain the correct credentials then rebooting, only then will the AP reappear.

ScottSturdivant, 2018-05-01:
Here is a workaround:

Edit the `/etc/dhcpcd.conf` file and for the `interface wlan1` line (changed due to running the AP on that interface), add in the `nohook wpa_supplicant` option.

With that change, the AP is visible after entering an incorrect client password.  I can then go back into the webui, change it to the correct password, and proceed.

Again, not sure this is a raspap-webgui bug, but wanted to leave the information here in case others are trying to do AP + Client with the RPi0W.  Thank you!

SirLagz, 2018-05-21:
Thanks for the info, we will look into this.

closed 2020-03-19 by stale[bot]