RaspAP/raspap-webgui #191: Use RaspAP as WiFi extender

monkiki, 2018-05-02:
### Subject of the issue
I want to use RaspAP as WiFi extender and not sure if it's possible.

### Your environment
* Raspberry Pi: Pi 3 Model B
* Raspbian: Stretch
* RaspAP: Quick Install

### Steps to reproduce
I've installed RaspAP and works fine with eth0 and wlan0, but because I want to extend my WiFi signal I've removed the eth0 plug and added a WiFi dongle. I've activated the IP Forwarding and executed these Iptables sentences:

```
iptables -F
iptables -X
iptables -Z
iptables -t nat -F

iptables -P INPUT ACCEPT
iptables -P OUTPUT ACCEPT
iptables -P FORWARD ACCEPT

iptables -t nat -A POSTROUTING -o wlan1 -j MASQUERADE
iptables -A FORWARD -i wlan1 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
iptables -A FORWARD -i wlan0 -o wlan1 -j ACCEPT
```

The **wlan1** is the built-in WiFi and is connected to the router's WiFi and the **wlan0** is the dongle and where **hostapd** and **dnsmask** are listening.

**/etc/dnsmasq.conf**
```
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,255.255.255.0,12h
```

**/etc/hostapd/hostapd.conf**
```
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
beacon_int=100
auth_algs=1
wpa_key_mgmt=WPA-PSK
ssid=raspi-webgui
channel=1
hw_mode=g
wpa_passphrase=ChangeMe
interface=wlan0
wpa=1
wpa_pairwise=TKIP
country_code=ES
```

### Expected behaviour
Raspberry should work as a AP and give Internet access to the clients.

### Actual behaviour
The AP partially works because I can see the SSID in the mobile, but once connected it sais "No Internet connection" :-(

SirLagz, 2018-05-21:
Yes it's possible. What you've posted seems like it's correct. Can the Raspberry Pi access the internet?

classofoutcasts, 2018-06-10:
Is ip forwarding eneabled?
```$ cat /proc/sys/net/ipv4/ip_forward```
should return a value of 1 
if not, try:
```echo "1" > /proc/sys/net/ipv4/ip_forward```
and see if it works.

galaxyfreak, 2018-06-22:
I want to achieve same thing but with only in-built WiFi

flatsiedatsie, 2018-08-07:
Have a look here:
https://github.com/lukicdarkoo/rpi-wifi

after reading this:
https://blog.thewalr.us/2017/09/26/raspberry-pi-zero-w-simultaneous-ap-and-managed-mode-wifi/

closed 2019-03-04 by billz