RaspAP/raspap-webgui #192: Enabling WPS

patricerosay, 2018-05-10:
### Subject of the issue (enhancement proposal)
Everything works fine. Great job!
How could I allow an WPS ready headless device to connect to the hotspot I've created ?
Could an additional page do that?

### Your environment
* Raspberry Pi  3 Model B,
* Raspbian version Stretch
* RaspAP Quick Install 

### Expected behaviour
startup the rpi and raspap-webgui
startup my device 
ask my device to lookup for a wifi connection
the new device is listed in raspap-webgui
I click on the device in raspap-webgui
the device has access to the hotspot

billz, 2018-10-14:
Nice to have but can't make any promises. Tracked with enhancement label. Closing for now, thanks

closed 2018-10-14 by billz

freezeeedos, 2021-11-03:
Hello,

I have setup wps based on this https://android.googlesource.com/platform/external/wpa_supplicant_8/+/499d68402f55bd1b8aa1a0d9c6966c7392704fba/hostapd/README-WPS

I wrote the following script:

```
#!/bin/bash

touch /etc/hostapd.psk
cp /etc/hostapd/hostapd.conf /etc/hostapd/hostapd.conf.bak
cat >> /etc/hostapd/hostapd.conf << EOF
wpa_psk_file=/etc/hostapd.psk
ctrl_interface=/var/run/hostapd
eap_server=1
wps_state=2
ap_setup_locked=1
wps_pin_requests=/var/run/hostapd.pin-req
device_type=6-0050F204-1
config_methods=label display push_button keypad
EOF

systemctl restart hostapd
```
Started wps enrollment with:

`sudo hostapd_cli wps_pbc`

And was able to connect my printer

I suppose this could be made available via the web GUI