RaspAP/raspap-webgui #193: wlan0 AP + eth0 internet not working

MohsinEngine, 2018-05-16:
The Access Point worked fine for first few times. I was able to access internet by connecting to the hotspot. But now hotspot fails to load up.

### Environment:
* Raspberry Pi 3 Model B  
* Raspbian Stretch **with Desktop**
* RaspAP Quick Install
* No other software installed

### Steps to reproduce:
1) Install fresh Raspbian Stretch with Desktop onto 16GB card.
2) after apt-get update and upgrade.
3) Edit **/etc/wpa_supplicant/wpa_supplicant.conf**. Delete all contents and replace with this:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```
4) Run this command to quick install RaspAP: 
`wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap`
5) Attach ethernet cable to the pi and router directly.
6) Reboot
7) Use laptop to connect to the Wifi hotspot if visible.
8) Reboot again
9) Hotspot won't be visible

### Another variation:
1) Remove ethernet cable.
2) Reboot pi
3) When it comes back up, the hotspot won't be visible.

### Expected behaviour
Hotspot should've been visible after reboot for devices to connect to

### Actual behaviour
Hotspot not visible anymore.

### Log file:
I enabled logging from the Web interface and it is as follow:
```
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:0f:87:dd and ssid "raspi-webgui"
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Only 0/20 bytes of strong random data available from /dev/random
random: Not enough entropy pool available for secure operations
WPA: Not enough entropy in random pool for secure operations - update keys later when the first station connects
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Cannot read from /dev/random: Resource temporarily unavailable
```

billz, 2018-06-06:
try:
`sudo rpi-update` to get the latest kernel and firmware
`sudo apt-get install rng-tools` to install tools for random number generation in the kernel

billz, 2019-03-13:
Best advice I can offer is to run the latest Quick Installer. There have been many updates since May 2018 when this issue was first posted. Also, be sure to follow the prerequisites in the readme as these steps can help avoid common problems down the road.

closed 2019-03-13 by billz