RaspAP/raspap-webgui #194: Is it possible to configure an alternate port for the webserver to access rasap?

ksmreddy, 2018-05-16:
### Alternate port for the webserver
How to change the access-port for Raspap-webgui instead of the default port-80?

### Your environment
* Raspberry Pi hardware:  Pi Zero W  
* Raspbian version: Stretch Lite
* RaspAP Quick Install 
* Other software or services running with RaspAP: ssh, 

### Expected behaviour
Something like <ip address of raspi0W running raspap>:xxxx.
Also any suggestion on the safe port numbers to use?

### Actual behaviour
currently raspap accessed on default port-80, just by typing the IP/name of raspi0W in the web-browser.

billz, 2018-05-23:
The default port is easily changed by editing `/etc/lighttpd/lighttpd.conf` like so:
```
server.port = 8080
```
Give the service a kick with `sudo lighttpd restart`

ksmreddy, 2018-05-23:
Thanks for the reply.

closed 2018-05-23 by ksmreddy

heatvent, 2019-12-21:
Maybe it would be good to be able to configure this when doing initial install. I would think many people are using RaspAP along with something else that has a webserver.  Just a thought.

billz, 2019-12-21:
@heatvent this would be a nice option for the installer. lighttpd's port can be configured via the UI now, but only after accessing it on port 80. thanks for the suggestion.