RaspAP/raspap-webgui #196: Use wlan1 as wlan client

Bigdesaster, 2018-05-16:
### Subject of the issue
Hello I'd like to use wlan1 as wlan client because i need an external antenna for the client. 
Wlan0 is configured as AP. 

### Your environment
* Raspberry Pi hardware: Pi 3B+
* Raspbian version:  Stretch
* RaspAP: Quick Install
* Other software or services running with RaspAP: OpenHAB

### Steps to reproduce
Goto WebIF section "ClientMode"

### Expected behaviour
It should be possible to select the interface which schould be configured as client

### Actual behaviour
 it's only possible to scan wlan0

Bigdesaster, 2018-05-23:
Thank you. I did allready but it doesn't work yet.  I found that the right vor wpa_cli is missing in sudoers for interfaces. My workaround is with wildcard for wlan[0-9]
But that's not all. I want to use wlan0 as hotspot and wlan1 as client.
 I'm currently trying to get it work. I will tell you here

Bigdesaster, 2018-05-23:
hello billz,
i think i got a quick workaround now:
with visudo i changed some permissions. In `install\common.sh` that should be:
`www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan[0-9] scan
www-data ALL=(ALL) NOPASSWD:	/sbin/ip link set wlan[0-9]
www-data ALL=(ALL) NOPASSWD:	/bin/cat /etc/wpa_supplicant/wpa_supplicant-wlan[0-9].conf
www-data ALL=(ALL) NOPASSWD:	/bin/cp /tmp/wifidata /etc/wpa_supplicant/wpa_supplicant-wlan[0-9].conf
` But maybe there should be used other wildcards because of compatinility to 'predictable network interface name'

I needed to change 2 lines in `/includes/config.php` and added another one:
`define('RASPI_WIFI_CLIENT_INTERFACE', 'wlan**1**');
define('RASPI_WPA_SUPPLICANT_CONFIG', '/etc/wpa_supplicant/wpa_supplicant**-wlan1**.conf');
**define('RASPI_WIFI_COUNTRY_CODE','DE');**`

In `/includes/configure_client.php` i added the line (56):
` fwrite($wpa_file, 'country=' . RASPI_WIFI_COUNTRY_CODE . PHP_EOL);` to add the country code.
This will be save into wpa_supplicant-wlan1.conf now, but maybe you should write the lines without the network {} part to `wpa_supplicant.conf` too.

Now i can acces my accespoint (wlan0) and connect to a client with wlan1. Unfortunately the routing from wlan0 to wlan1 does not work yet, but at the moment that's not part of your project.
Maybe it will be one time, i hope so. ;). 
Now im going to hollidays. Maybe i can help to improve your project if you like that after my holidays. Just let me know.

gherlein, 2018-05-27:
I want this exact functionality as well.  I'll take a peek and maybe send a pull request.
I'm also wondering how this can be templatized so that on install the key variables (SSID, IP address, Password) could be optionally specified to simplify the steps for an install - in other words, to configure the AP without even needing to use the GUI even that first time

gherlein, 2018-05-27:
And Country code.  I'm imagining a file that could be dropped onto the /boot partition of the SD card before first boot that would "just configure" this all to install and configure on first boot and still enable the interface after.

gherlein, 2018-05-27:
However, this article:  https://medium.com/poka-techblog/diy-wifi-router-using-kali-linux-on-a-raspberry-pi-3-model-b-89b8e5497cbf says that the BCM2837 is unstable in master mode, which matches my experiences this morning playing around with it.  I wonder if shifting to wlan1 for the AP using another adapter may be a better overall option.

randyoo, 2018-07-06:
Just found this issue. I had already issued a pull request adding wlan1 to /etc/sudoers, but your solution looks nicer. Predictable interface name support looks like a nightmare, though!

@gherlein : I had lots of trouble in the past with stability of the internal WLAN in AP mode, but they have been working on the driver, and my recent experience is that it's been perfectly stable for nearly a week.

tekzoom, 2018-07-31:
Hello @Bigdesaster,
I am not sure what I missed here. I was going to add the wlan1 (usb dongle) as wifi client but no success so far!
Would you please indicate which permissions you changed with visudo?
In the steps you described above, did you change install\common.sh, /includes/config.php, and configure_client.php before or after installing the raspap?

Dude4Linux, 2019-03-16:
@gherlein - I've experienced problems keeping `hostapd` running on my Rpi 3B.  I recently discovered that it has something to do with `TKIP`.  After switching to `CCMP`, it has been working well.

randyoo, 2019-03-25:
> @gherlein - I've experienced problems keeping `hostapd` running on my Rpi 3B. I recently discovered that it has something to do with `TKIP`. After switching to `CCMP`, it has been working well.

[I had the same experience.](https://github.com/raspberrypi/linux/issues/1342#issuecomment-351501303)

billz, 2019-04-09:
> I'm also wondering how this can be templatized so that on install the key variables (SSID, IP address, Password) could be optionally specified to simplify the steps for an install - in other words, to configure the AP without even needing to use the GUI even that first time

The next step for this project will be using JSON for config management. Stay tuned

billz, 2019-04-13:
@tekzoom I've added @randyoo's input to our FAQ, please check it out.

Closing this issue for now, feel free to update if you have more info. Thanks

closed 2019-04-13 by billz