RaspAP/raspap-webgui #198: Using wlan0 and wlan1 rather than eth0

schwieb, 2018-06-05:
I'd like to use the built in wlan0 connection and a usb wlan1 connection rather than eth

* Raspberry Pi hardware: Pi 3 Model B
* Raspbian version: Stretch
* RaspAP Quick Install 

Unfortunately, every time I connect wlan0 to the local AP, it also connects the wlan1 to that same AP rather than broadcasting. How do I dedicate one of the wlan devices to the WAN Internet and the other to the LAN?

Thanks!

randyoo, 2018-07-06:
I had the same problem, and I managed to resolve it by adding `nohook wpa_supplicant` to the interface in the `/etc/dhcpcd.conf` file.
I have wlan0 serving as the access point, and wlan1 as client. I'm not 100% sure what was there originally, but in my current, working configuration, the following lines to the bottom of my `/etc/dhcpcd.conf`:
`interface wlan0   
        static ip_address=10.3.141.1/24
        static domain_name_servers=8.8.8.8
        nohook wpa_supplicant`

Beware, there are a number of other changes that will need to be made to get this working. For example, in `/var/www/html/includes/config.php`, you'll need to change `define('RASPI_WIFI_CLIENT_INTERFACE', 'wlan0');` to wlan1.

You'll also need to add the following lines to the `/etc/sudoers` file, if you want scan functionality to work in RaspAP:
`www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan1 scan_results
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli -i wlan1 scan`

billz, 2019-04-13:
@randyoo I've updated the wiki accordingly, thanks

closed 2019-04-13 by billz

lehni, 2020-06-05:
@billz the instructions by @randyoo work, but as soon as I change anything else in RaspAP, the `/etc/dhcpcd.conf` file gets overridden again, and it looks like RaspAP sets the wrong wlan name then aslo, using `wlan1` instead of `wlan0`:

```
interface wlan1
static ip_address=10.3.141.1/24
static routers=10.3.141.1
static domain_name_server=1.1.1.1 8.8.8.8
```

Sadly, this makes using RaspAP not very practical for such configurations.

Also, note that `static routers=10.3.141.1` needs to be removed / disabled, as otherwise the route through wlan1 does not actually work.

billz, 2020-06-05:
@lehni there is an active PR that adds better support for switching the AP interface. Perhaps you could contribute to it.

lehni, 2020-06-06:
@billz I was looking through the issues but didn't look at the PRs. I'm happy to help with #582. What is left to be done?