RaspAP/raspap-webgui #200: Hotspot not starting after wifi configuration

aaamitsingh, 2018-06-20:
### Subject of the issue
Describe your issue here.

### Your environment
* Raspberry Pi hardware (Pi 3 Model B+)  
* Raspbian version (VERSION="9 (stretch))
* RaspAP Quick Install.
* No other software or services running with RaspAP.

### Steps to reproduce
*Install fresh Raspbian Stretch with Desktop onto 16GB card.
after apt-get update and upgrade.
*Run this command to quick install RaspAP: wget -q https://git.io/voEUQ -O /tmp/raspap && bash /tmp/raspap
* Stop the hostapd.
* Configure the wifi with RaspAP then reboot the Pi
* Reeboot the Pi
* Wifi is Up.
* Again tried to Start hotspot but its not starting.
* Tried this command
sudo rpi-update
sudo apt-get install rng-tools
* Both above two commands executed successfully
* Reboot the pi again. 

### Expected behaviour
Hotspot is not starting

### Actual behaviour
This is my log:--
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: STA 70:4f:57:9d:fa:82 IEEE 802.11: disassociated
wlan0: STA 70:4f:57:9d:fa:82 IEEE 802.11: disassociated
wlan0: STA 70:4f:57:9d:fa:82 IEEE 802.11: disassociated
wlan0: STA 70:4f:57:9d:fa:82 IEEE 802.11: disassociated
wlan0: STA 70:4f:57:9d:fa:82 IEEE 802.11: disassociated
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:ef:d4:35 and ssid "raspi-webgui"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED 
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0

FredLeudon, 2018-06-29:
Don't know if Your issue is the same as mine but on raspberry pi 2 with official WiFi dongle, hostapd service won't start.

I updated /etc/init.d/hostapd like this : 
in start section, added command `killall wpa_supplicant`

` start)`
`       log_daemon_msg "Starting $DESC" "$NAME"`
`       killall wpa_supplicant`
`       start-stop-daemon --start --oknodo --quiet --exec "$DAEMON_SBIN" \`
`        --pidfile "$PIDFILE" -- $DAEMON_OPTS >/dev/null`
`        log_end_msg "$?"`
`        ;;`

alpgul, 2018-08-25:
i have same problem.

Offbeatmammal, 2018-08-30:
seeing the same problem with current Raspbian/RPi B+

Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:b1:cb:7f and ssid "obmpi"
wlan0: interface state COUNTRY_UPDATE->ENABLED
wlan0: AP-ENABLED 

it sometimes works, but not consistently.

closed 2019-04-13 by billz