RaspAP/raspap-webgui #202: Tested on Stretch and RasPi 3B+?

glynhudson, 2018-06-26:
Has anyone been able to successfully run this on a RasPi 3B+ running Stretch?

BI7PRK, 2018-07-09:
yes, i can run on pi3b+  kali OS. but it can't authenticate.  I have tried all the encryptions.

glynhudson, 2018-07-18:
What do you mean by can't authenticate? I've just tried running on RasPi 3B+ (default Raspbian Stretch). It seems to startup ok, however, hotspot doesnt seem to work and it doesn't seem to be able to scan for client networks. Anyone else tried using this with Stretch? Any additional steps required?

ginodouglas, 2018-07-29:
I had the same problem with pi3b+ and Stretch, putting the SD card in a pi3b it worked fine.

billz, 2018-09-07:
Did you set up everything in raspi-config? The WiFi country setting is important. 

Per the Raspbian [release notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt), WiFi is disabled until the wireless regulatory domain is set (Pi 3 B+ only). There is no default value. The quick installer doesn't handle this step for you.

Be sure to shutdown and boot (not reboot) your Pi  after changing this setting.

If you have trouble getting the hostapd service to start, add the following to `/etc/rc.local` just before `exit 0`:

```
service hostapd stop
sleep 5
service hostapd start
```

danielwilson654, 2018-11-12:
I have the same issue on a clean install of ubuntu 18.04 on arm. 
```
Attempting to start hotspotx
HostAPD is not running
```
Here is the output of systemctl status hostapd

`● hostapd.service - LSB: Advanced IEEE 802.11 management daemon
   Loaded: loaded (/etc/init.d/hostapd; generated)
   Active: active (exited) since Mon 2018-11-12 03:42:44 UTC; 10min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 1395 ExecStop=/etc/init.d/hostapd stop (code=exited, status=0/SUCCESS)
  Process: 1416 ExecStart=/etc/init.d/hostapd start (code=exited, status=0/SUCCESS)

Nov 12 03:42:44 nanopiair systemd[1]: Starting LSB: Advanced IEEE 802.11 management daemon...
Nov 12 03:42:44 nanopiair systemd[1]: Started LSB: Advanced IEEE 802.11 management daemon.
`

danielwilson654, 2018-11-12:
This might be the case as ubuntu doesn’t use a wpa_supplicant.conf file. On ubuntu, wpa_supplicant it is managed dynamically somehow.

billz, 2018-11-12:
>  On ubuntu, wpa_supplicant it is managed dynamically somehow. 

For better or worse, Ubuntu 18.04 now uses `netplan` to configure networking. As RaspAP is designed for Raspbian, this setup is not supported.

billz, 2019-03-04:
Please refer to https://github.com/billz/raspap-webgui/issues/287#issuecomment-469083944

Closing this issue. Reopen if needed, thanks.

closed 2019-03-04 by billz