RaspAP/raspap-webgui #206: Can't access any website except www.google.com

dead-alchemist, 2018-07-06:
After installing Rasp-AP on my Raspberry Pi 3 Model B "Jessie" , everything works perfectly fine but i am not able to access any website except  google.com . I have gone through a simple quick install setup and nothing seems wrong while the processes keep on executing. 
Help!!!

billz, 2018-10-14:
From the shell, edit `/etc/network/interfaces` and update like so: 
```
dns-nameservers 1.1.1.1
```
Then give the service a kick:
```
sudo service networking restart
```
If you still have DNS issues, your original network config is backed up to `/etc/raspap/backups/interfaces`. You can either restore your settings from here or run the provided uninstaller to cleanly return your RPi to its prior state.

billz, 2019-01-23:
Closing due to inactivity. Feel free to reopen if your issue persists, thanks

closed 2019-01-23 by billz