RaspAP/raspap-webgui #208: Cant create Open wifi - not Secured

CarloCattano, 2018-07-13:
I will like to create an open Wifi , without the need of password

Pi 3 Model B,

mpkogli, 2018-07-24:
For whatever reason, RaspAP does not allow this. But you can manually edit your /etc/hostapd/hostapd.conf file as follows, which works for me with a Rasp B+ and the standard Edimax adapter:

```
interface=wlan0
driver=nl80211
ssid=NotFreeWiFi
channel=1
```

Note that I had to use wlan0 as the host AP interface and wlan1 as the client. This was accomplished by renaming the /etc/wpa_supplicant/wpa_supplicant file to /etc/wpa_supplicant/wpa_supplicant-wlan1.conf. Additional changes are necessary in the RaspAP code to make this work fully. See https://github.com/billz/raspap-webgui/issues/198 and https://github.com/billz/raspap-webgui/issues/196 for more info.

Ideally this could be rewritten to allow specific configuration during installation, but it's working for me now :)

flatsiedatsie, 2018-08-07:
Just put something like "password = iloveyou" in the SSID name. Then people can still use it as a public access point without losing security.

D9ping, 2018-09-13:
For now, don't use a = sign in the SSID name. 
I have seen there is no escaping of configuration values yet.
The /etc/hostapd/hostapd.conf file can become invalid making it impossible to start the hotspot.

mathiasconradt, 2018-10-13:
The linked commit by @D9ping above is not complete yet, while it shows the option "None", it does not let you save because the form validation still requires a certain password length.