RaspAP/raspap-webgui #209: wireless mode n only

DiachenkoIgor, 2018-07-19:
### Subject of the issue
There is no n wireless mode. How can i turn it on?

### Your environment
* Raspberry Pi hardware (Pi 3 Model B+)  
* Raspbian version (Raspbian)
* RaspAP Quick Install or Manual setup(Quick)?
* Other software or services running with RaspAP(ffmpeg)?

### Steps to reproduce
Tell us how to reproduce this issue.

### Expected behaviour
Tell us what should happen.

### Actual behaviour
Tell us what happens instead.

D9ping, 2018-08-02:
As workround.
Connect via ssh with an ethernet cable to the pi 
or open a terminal with a display and a keyboard connected to the pi.

Execute:
sudo -s
Type in your pi root account password blindly and press enter.

Run:
echo "ieee80211n=1" >> /etc/hostapd/hostapd.conf

And then run(not via wi-fi, as connection gets lost as hotspot is restarted), to restart the hotspot with 802.11 N support:
systemctl restart hostapd.service

billz, 2018-10-06:
This is now supported in the UI by @D9ping's latest PR. Thanks for reporting

closed 2018-10-06 by billz