RaspAP/raspap-webgui #214: HostAPD not running

alanionita, 2018-07-26:
### Subject of the issue
Trying to use RaspAp to create a local hotspot but it's not discoverable. I attempted a purely wifi setup and a Ethernet connection with similar results.

The Pi is a fresh install. 

### Your environment
* Hardware: Pi 3 Model B
* Raspbian version: Stretch
* RaspAP Quick Install
* No other software

### Steps to reproduce
- connected to local wifi
- clean install of Raspbian + update
- quick install of RaspAP
- start hotspot on Wifi,
- plug in Ethernet and reboot
- start hotspot,

### Expected behavior
- I was hoping to have the configured hotspot appear in the available network list
- after connecting via Ethernet I was hoping I could restart the hotspot and it would be discoverable

### Actual behavior
- the hotspot turned on but the was not discoverable
- the hotspot failed to start with the HostAPD is not running error

### Log
```bash
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
random: Cannot read from /dev/random: Resource temporarily unavailable
wlan0: IEEE 802.11 Configured channel (1) not found from the channel list of current mode (2) IEEE 802.11a
wlan0: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

alanionita, 2018-07-27:
Solution:

- add ```killall wpa_supplicant``` command to the start method of **/etc/init.d/hostapd**
```
start)
     log_daemon_msg "Starting $DESC" "$NAME"
     // ADD THE BELOW LINE
     killall wpa_supplicant
     start-stop-daemon --start --oknodo --quiet --exec "$DAEMON_SBIN" \
     --pidfile "$PIDFILE" -- $DAEMON_OPTS >/dev/null
     log_end_msg "$?"
    ;;
```
- update Pi firmware using ```sudo rpi-update```

closed 2018-07-27 by alanionita

enseitankado, 2019-01-28:
The solution (adding killall wpa_supplicant command to the start method of /etc/init.d/hostapd) solved my issue. 

Please include the line to webgui.

Thanks.

NilsonCCL, 2019-02-12:
I have the same problem!
When starting manually it works but by RaspAP it does not work!
Still no solution!