RaspAP/raspap-webgui #216: Cannot access the Internet over wlan0 when RaspAP is installed

HelloOO7, 2018-07-31:
### Subject of the issue

When RaspAP is installed, I cannot access the Internet over wlan0. Hostapd is stopped, of course, the Raspbian wifi status bar icon says that:

wlan0: associated with <my home wifi SSID>

however,

wlan0: Configured 10.3.141.1/24

(instead of 192.168.1.xx)

All webpages accessed at this state return "cannot resolve dns". Without RaspAP, the wifi connection works fine.

### Your environment
* Raspberry Pi hardware: Pi 3 Model B
* Raspbian version: Retropie Stretch
* RaspAP Quick Install

### Steps to reproduce
1. Flash a Raspbian Stretch image (retropie in my case) and install the Pixel desktop environment
2. Download and run the quick install script (no errors occured)
3. Reboot
>RaspAP should be active
>wlan0 cannot acces the internet

### Expected behaviour; (some words in brackets)==actual behaviour

I should be able to use the hostapd hotspot (works) and when it is off, access the Internet over wlan0 (doesn't work).

jeijunn, 2018-08-24:
In my case, I wanted to use raspberry pi to share its wireless connection by ethernet cable.
So what I did was as following. 

Access your raspap web page by running a web browser from raspberry pi :
http://127.0.0.1  

Then log in with default id/password :
admin/secret 

Go to Configure DHCP Server and edit your settings :
Change Interface : wlan to eth0, then save settings, stop dnsmasq then start it again.

Go to Configure hotspot and edit :
Basic settings->Interface : wlan to eth0, save settings, stop hotspot and start again.

billz, 2019-04-09:
Premade third-party images are generally unsupported. Having said that, it sounds like Retropie uses settings that conflict with RaspAP's default configuration. In this case a manual installation will probably work better for you. 

Feel free to update this issue with more info. Thanks.

closed 2019-04-09 by billz