RaspAP/raspap-webgui #221: Cannot scan for networks "Undefined variable: networks"

glynhudson, 2018-08-07:
### Subject of the issue
The GUI is not able to scan for networosk 

### Your environment
* Raspberry Pi hardware: 3B+
* Raspbian version: Stretch
* RaspAP Quick Install or Manual setup?: Manual 
* Other software or services running with RaspAP?: Moved lighttpd port 81 since apache is running Emoncms on port 80

### Steps to reproduce
Install and run 

### Expected behaviour
Local WiFi networks should be displayed

### Actual behaviour

[image]

```
$ sudo  tail  /var/log/lighttpd/error.log
2018-08-07 17:48:27: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Notice:  Undefined variable: networks in /var/www/html/includes/configure_client.php on line 183
2018-08-07 17:48:27: (mod_fastcgi.c.2543) FastCGI-stderr: PHP Warning:  Invalid argument supplied for foreach() in /var/www/html/includes/configure_client.php on line 183
```

mlekorlz, 2018-08-21:
I have the same problem with Pi Zero W

D9ping, 2018-10-12:
Could you run on a terminal:
``
sudo wpa_cli -i wlan0 scan
``
and then run:
``
sudo wpa_cli -i wlan0 scan_results
``

And tell me if you then see a list of wi-fi networks on the terminal.

PRO2XY, 2018-11-03:
@D9ping @glynhudson : Please see [my reply on issue 245](https://github.com/billz/raspap-webgui/issues/245#issuecomment-435588181).
This is due to incomplete entry on `/etc/sudoers` for `wpa_cli` it needs the interface to be defined in sudoers file, or the entire `wpa_cli` exposed.
Another problem is the wrongly defined variable `$scanned_networks` which needs to be `$networks`.

Please correct me if I am wrong.

billz, 2019-03-05:
Resolved with https://github.com/billz/raspap-webgui/commit/d912eb5a1272d769bbe1fbfd365af7ae0228e215.

closed 2019-03-05 by billz