RaspAP/raspap-webgui #226: Ambiguous license terms suggest charges for using RaspAp and its embedded font-awesome

clarkek, 2018-08-31:
### Subject of the issue
License for fontawesome appears to be commercial with fees, and this is not clearly stated, while there are free versions of fontawesome, so some clarity would be appreciated.

### Your environment
* Raspberry Pi 3 Model B+ 
* Raspbian Stretch, 
* Manual setup

### Steps to reproduce
Reading  license for the fontawesome package files.
vendor/font-awesome/css/font-awesome.css 

Line in file says

 *  License - http://fontawesome.io/license (Font: SIL OFL 1.1, CSS: MIT License)

However the page linked to only mentions commercial terms, while showing links to free versions on fortawesome 

*The Company develops Font Awesome Free, a set of icons, related computer software, and other material published at github.com/FortAwesome/Font-Awesome and fontawesome.com. This license is for Font Awesome Pro, a suite of additional Pro Icons and Pro Software:*

### Expected behaviour

There should not be such ambiguity in the license header and terms and should state more clearly that there is chargeable licenses implied by using font awesome and therefore also by raspap.

OR
it should clearly be and state that is font awesome free with its no charges

OR
it should provided a method to switch to the free licensed versions.

### Actual behaviour
The license terms imply there is commercial license, while still saying they are MIT .

billz, 2018-10-14:
RaspAP is built with SB Admin 2, which uses the free [Font-Awesome 4.6.3](https://github.com/FortAwesome/Font-Awesome/releases/tag/v4.6.3) release dated May 13, 2016. You can find this in the original [bower package](https://github.com/BlackrockDigital/startbootstrap-sb-admin-2/blob/master/bower.json) which this project also adopted. The Pro license was introduced with Font Awesome 5, released in December 2017. 

In this case, there was only one MIT license available (free) when Font-Awesome 4.6.3 was released. I suspect the Free license page was added after the release of Pro. The latter became the default MIT license page linked in the `font-awesome.css` file. Given that Bower simply builds packages from [their source components](https://github.com/FortAwesome/Font-Awesome/blob/v4.6.3/css/font-awesome.css), there's nothing to fix here.

clarkek, 2018-10-18:
That clarity and chain of inheritance clears it up. Thank You

closed 2018-10-18 by clarkek