RaspAP/raspap-webgui #227: Can this be used a a backup internet connection

TheCellMC, 2018-09-02:
I've got a quick question. Would it be possible to use this project as a backup internet connection.
For example, if my internet goes down, this kicks in and provided internet via the 4g dongle until the main internet it back online ?

billz, 2018-09-02:
Absolutely. @SirLagz has [an excellent walkthrough](http://sirlagz.net/2017/10/22/using-my-raspberry-pi-as-a-backup-internet-connection/) on this topic.

TheCellMC, 2018-09-03:
I believe that that blog port doesn't use this project. He mentions that he might use this project as a future upgrade.
Is it possible to use this project to achieve this entirely via the UI ?

TheCellMC, 2018-09-03:
Thanks for your help @billz and at @SirLagz

closed 2018-09-03 by TheCellMC