RaspAP/raspap-webgui #228: Proposed enhancement - show bandwidth statistics per day, week, month

D9ping, 2018-09-04:
### Show bandwidth use of rasberry pi per day, week, month
This is a optional page that shows the output of vnstat to display the bandwidth use of RaspAP 
 per day, per week or per month, for a selected interface.
If can be enabled just like openvpn and tor.

D9ping, 2018-09-04:
I already have a branch ready i can turn into a PR.
The impact on the code is minor, as it introduces mainly a new file for the page to display(vnstat.php) and 
edits the menu bar(index.php) and new config constant: RASPI_VNSTAT_ENABLED.

billz, 2018-09-07:
Really nice addition @D9ping. I tested your feature branch and it's nearly there. I noticed the stats are mostly preformatted raw output of `vnstat`. For consistency, it'd be great if this were parsed and displayed in a responsive table, similar to the Wifi Configuration UI. Maybe you're already doing this? If not create a PR and I'll add to it.

D9ping, 2018-09-12:
Some notes about latest pull request.
- Charts and datatable are all loaded with ajax requests as tab or an interface is selected.
- As Morris.js and jquery.dataTables javascript depend on jquery, they get added to $extraFooterScripts (by reference) and then they be inserted in index.php after jquery has been inserted in the page.
- Bandwidth page is currently disable by default, as it otherwise makes vnstat a dependency to install RaspAP.

billz, 2018-09-13:
@D9ping outstanding addition to the project. I think it should be part of the default setup and enabled by default, with the option of disabling as with other services in `config.php`. Morris.js has been part of the SB-Admin theme from the start but we haven't used it (yet). This is an excellent application. 

vnstat is fairly lightweight and could be installed along with other dependencies in `raspbian.sh`

closed 2018-09-17 by D9ping