RaspAP/raspap-webgui #234: How do I make the hotspot activate on startup and after reboot?

tampadrones, 2018-09-13:
### Subject of the issue
I want the hotspot to be active when the pi boots or restarts

### Your environment
RasPi 3b+
Raspian Stretch
Apache

### Steps to reproduce
After setting up RaspAP I open the web based control panel and click Hotspot. I see that it shows as running. I click stop twice to get a screen confirming the hotspot is no longer running. I click Start Hotspot and my new network then shows. 

### Expected behaviour
I want the hotspot to be active on every boot, whether this is an entry in crontab or otherwise. 

### Actual behaviour
No wireless network is broadcasting unless stopped and started again in the control panel

tampadrones, 2018-09-13:
It seems like that if the pi boots with the keyboard, mouse and monitor, then it works as expected...

BANKA2017, 2018-10-09:
Open /etc/rc.local with a text editor and add this just above "exit 0":

>service dhcpcd stop
>sleep 5
>service hostapd stop
>sleep 5
>service hostapd start
>sleep 5
>service dhcpcd start

maybe can help u?

jdiegosierra, 2018-10-10:
> Open /etc/rc.local with a text editor and add this just above "exit 0":
> 
> > service dhcpcd stop
> > sleep 5
> > service hostapd stop
> > sleep 5
> > service hostapd start
> > sleep 5
> > service dhcpcd start
> 
> maybe can help u?

whith:
service hostapd start
it works

billz, 2019-02-07:
@tampadrones did the above fix resolve your issue? If not, please reopen and update with any findings. Thanks

closed 2019-02-07 by billz

andy-yeung, 2021-01-15:
> @tampadrones did the above fix resolve your issue? If not, please update with any findings. Thanks

system show: Unit raspapd.service could not be found.

how do i install raspapd.service?

billz, 2021-01-15:
@andy-yeung raspapd.service is added by the installer. You can run the installer again or add it manually:

```
sudo cp /var/www/html/installers/raspapd.service /lib/systemd/system/ 
sudo systemctl daemon-reload
sudo systemctl enable raspapd.service
```

andy-yeung, 2021-01-15:
> @andy-yeung raspapd.service is added by the installer. You can run the installer again or add it manually:
> 
> ```
> sudo cp /var/www/html/installers/raspapd.service /lib/systemd/system/ 
> sudo systemctl daemon-reload
> sudo systemctl enable raspapd.service
> ```

> @andy-yeung raspapd.service is added by the installer. You can run the installer again or add it manually:
> 
> ```
> sudo cp /var/www/html/installers/raspapd.service /lib/systemd/system/ 
> sudo systemctl daemon-reload
> sudo systemctl enable raspapd.service
> ```

oh!! Thanks!!!
May I ask another question?
I'm using WIFI client AP mode.
how do I automatically make the dnsmasq activate after reboot?

billz, 2021-01-16:
> how do I automatically make the dnsmasq activate after reboot?

The raspapd.service starts up hostapd, dhcpcd & dnsmasq on boot and has special handling for WiFi client AP mode.
In fact, it was originally created for this purpose 😉

andy-yeung, 2021-01-18:
> The raspapd.service starts up hostapd, dhcpcd & dnsmasq on boot and has special handling for WiFi client AP mode.
> In fact, it was originally created for this purpose 😉

I still have the error,it prompt 
Jan 18 09:35:24 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jan 18 09:35:24 raspberrypi dnsmasq[1105]: dnsmasq: syntax check OK.
Jan 18 09:35:24 raspberrypi dnsmasq[1107]: dnsmasq: failed to create listening socket for 192.168.50.1: Cannot assign requested address
Jan 18 09:35:24 raspberrypi systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Jan 18 09:35:24 raspberrypi systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Jan 18 09:35:24 raspberrypi systemd[1]: dnsmasq.service: Unit entered failed state.
Jan 18 09:35:24 raspberrypi systemd[1]: dnsmasq.service: Failed with result 'exit-code'.

but I manually type "sudo systemctl start dnsmasq" after reboot, the service gonna be active

andy-yeung, 2021-01-18:
> I still have the error,it prompt
> Jan 18 09:35:24 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
> Jan 18 09:35:24 raspberrypi dnsmasq[1105]: dnsmasq: syntax [...]

and I solved it after researching [https://www.raspberrypi.org/forums/viewtopic.php?t=215235]
Just remove the bind-interfaces.
Thank you so much for your great support agian! 😉😉