RaspAP/raspap-webgui #237: Unable to logon to FreePBX post installing RaspAP WebGui on Raspberry Pi 3B+.

gururajksa, 2018-09-17:
### Subject of the issue
Post installing RaspAP on to FreePBX unable to login to FreePBX Admin portal

Its re-directing to RaspAP-WebGUI

Secondly DHCP not allocating IP to WLAN0. It says DHCP services on all interface failed, see systemctl status dhcp.service

### Your environment
* Raspberry Pi 3 B+
### Steps to reproduce
1. Installed FrePBX 04-04-2018 from Raspberry URL
2. Installed RaspAP Wireless Router on the same

### Expected behavior

1. RaspAP WebGUI - DHCP should allocate IP to all Network Interfaces
2. FreePBX Admin page login for configuring and managing Asterisk PBX

### Actual behaviour
1. Unable to connect FrePBX Admin portal
2. DHCP service failed post edit of /etc/network/interfaces post adding wlan0 as per some forum

billz, 2018-09-17:
Installing RaspAP alongside FreePBX is not supported. The uninstaller will return your Pi to its prior state

closed 2018-09-17 by billz

gururajksa, 2018-09-18:
Hello Billz, any other way to have both Wireless Access Point and FreePBX on Raspberry Pi 3?
Please suggest we have a use case where we need to have FreePBX and Wireless AP on Raspberry Pi 3.