RaspAP/raspap-webgui #239: install with 2 wired ethernet

lebphi, 2018-09-23:
hi,

it is possible to install raspap with 2 ethernets connections. 

I would like to replace the wifi wlan0 by a wired ethernet eth1 (usb to ethernet adapter) ?

regards

philippe

billz, 2018-10-07:
RaspAP won't create a wired network bridge for you, but you can manage it after you do the manual setup. 

Install `bridge-utils` with `sudo apt-get install bridge-utils`. Assuming your USB adapter creates an interface called eth1, create a backup of `/etc/network/interfaces` and assign the interfaces like so:

```
auto lo
iface lo inet loopback
 
iface eth0 inet manual
 
iface eth1 inet manual
 
auto br0
iface br0 inet dhcp
  bridge_ports eth0 eth1
```
Reboot the RPi and you should see a **br0**  interface in the Configure Networking panel.

This issue is beyond the scope of the project, so closing it for now.

closed 2018-10-07 by billz

toshibochan, 2019-12-02:
I just did this to use usb Ethernet adapter.
https://github.com/billz/raspap-webgui/issues/446