RaspAP/raspap-webgui #240: Infinite Lease Interval causes dnsmasq to no longer start

leonhess, 2018-09-25:
### Subject of the issue
when selecting the infinite lease interval, while still having a number in the ``lease time`` field causes the dnsmasq sevice to no longer start since the ``/etc/dnsmasq.conf`` is then wrongly configured.

### Your environment
* Raspberry Pi hardware: Pi 3 B+
* Raspbian version: newest
* RaspAP Quick Install

### Steps to reproduce
- Go to ``Configure DHCP Server`` and there to ``Server Settings`` in the web interface
- Set ``Interval`` to Infinite and leave a number in ``Lease Time``, i.e. ``13``

### Expected behaviour
Dnsmasq should start up with an inifinite lease time and the config should look like below.

#### /etc/dnsmasq.conf:
````
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,255.255.255.0,infinite
````
### Actual behaviour
Dnsmasq fails to start, since the config looks like below.
#### /etc/dnsmasq.conf:
````
interface=wlan0
dhcp-range=10.3.141.50,10.3.141.255,255.255.255.0,13infinite
````
Manually Deleting the 13 afterwards or not entering a number in the ``lease time`` field fixes this. But it took me a while to figure out where the problem was. Maybe blocking the ``lease time`` field when ``infinite```is selected would be a great addition.

billz, 2018-09-26:
@leonhess I merged a PR from @D9ping which fixes this issue. Thanks

billz, 2018-10-02:
Closing due to inactivity. Feel free to reopen if this doesn't resolve your issue

closed 2018-10-02 by billz